felHR85/UsbSerial #3: main program for communication usb serial

alfakr, 2015-01-04:
hallo felHR85,
first of all thank you for your usb serial library, 
i would ask about main program for simple read and write via usb serial with using your library, (Im using Prolific usb-to-serial  PL2303) 
I newby on developing android so its hard enough for me to start writing code,, 
this is a code I write, and yes there is an error when I click button, its force closed, 
hope you have a little time for helping me, 
Thanks

package com.android.testcomserial;

import android.app.Activity;
import android.hardware.usb.UsbDevice;
import android.hardware.usb.UsbDeviceConnection;
import android.os.Bundle;
import android.util.Log;
import android.view.View;
import android.widget.Button;
import android.widget.Toast;

import com.felhr.usbserial.PL2303SerialDevice;
import com.felhr.usbserial.UsbSerialInterface;

public class MainActivity extends Activity {
	
	//UsbManager usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
	private static UsbDevice device;
	private static UsbDeviceConnection usbConnection;
	private final PL2303SerialDevice pl2303 = new PL2303SerialDevice(device, usbConnection); 
	
	Button buttonWrite;
        
        @Override
	    protected void onPause() {
	        super.onPause();
	        if (pl2303 != null) {
	            try {
	            	pl2303.close();
	            } catch (Exception e) {
	                Log.e("LOG", e.toString());
	            }
	        }
	        finish();
	    }

	 @Override
	 protected void onResume() {
	        super.onResume();
	        if (pl2303 == null) {
	            Toast.makeText(getApplicationContext(), "No serial device.", Toast.LENGTH_LONG).show();
	        } else {
	            try {
	            	pl2303.open();
	            	pl2303.setBaudRate(115200);
	            	pl2303.setDataBits(UsbSerialInterface.DATA_BITS_8);
	            	pl2303.setParity(UsbSerialInterface.PARITY_NONE);
	            } catch (Exception e) {
	                Log.e("Error setting up device: ", e.toString());
	                //pl2303.close();
	            }
	        }		
	}
	
	 
	@Override
	protected void onCreate(Bundle savedInstanceState) {
		super.onCreate(savedInstanceState);
		setContentView(R.layout.activity_main);
		buttonWrite = (Button)findViewById(R.id.buttonWrite);
		buttonWrite.setOnClickListener(new View.OnClickListener() {
			@Override
			public void onClick(View v) {
				// TODO Auto-generated method stub
				pl2303.write("DATA FROM ANDROID".getBytes());
				
			}
		});
		
	}
	
	 private UsbSerialInterface.UsbReadCallback mCallback = new UsbSerialInterface.UsbReadCallback() {

        @Override
        public void onReceivedData(byte[] arg0) 
        {
            // Code here :) // what should I write here??
        	Toast.makeText(getApplicationContext(), "new data receive", Toast.LENGTH_LONG).show();
        }
	 };
}

felHR85, 2015-01-05:
I am going to keep open this sometime. But it does not look a library's problem just a problem with your Android code. I sent you some advices by email.

alfakr, 2015-01-06:
@felHR85 , yes the problem is not on your code but on me, 
btw really big thanks for your help, i was reveiced your email and I will try that,,,
I will inform you asap

closed 2015-01-30 by felHR85

alfakr, 2015-01-30:
oke,
thanks for your helping,, :)

Dani

eslovenit, 2015-05-14:
Hello , I am doing the same.. can you give me advices please?