felHR85/UsbSerial #6: arduino controlling

IlymH, 2015-03-09:
hi; I just wanted if there is any example on how to recieve data from socket and send it to aruino then do the reverse thing, I have a very vry little experience with android, but I did it in pc with rwtx.. library; so I'll be so thankful to you.

felHR85, 2015-03-09:
Hi @IlymH 
There is one example available of a very simple but complete application using UsbSerial (Pay close attention to the UsbService.java file):
https://github.com/felHR85/SerialPortExample

I would recommend reading this post first: 
http://felhr85.net/2014/11/11/usbserial-a-serial-port-driver-library-for-android-v2-0/

This shouldn't be no problem with an Arduino. As far as I know they use a FT232 and those are fully supported.

Hope it helps and keep me updated it it works!

IlymH, 2015-03-09:
thank you sir, I'll try it

pieman64, 2015-03-13:
Hi Felipe,
Your SerialPortExample is working fine with the Prolific Pl2303HX to an Arduino. But direct to the Arduino (without the Prolific) it starts up with USB found and then USB not supported. Are you saying direct to the Android should be ok?

felHR85, 2015-03-13:
Hi @pieman64 
It should be working. What arduino are you using it?

pieman64, 2015-03-13:
Nano clones Felipe.
USB\VID_1A86&PID_7523&REV_0254
I have a Mega and a Leonardo as well so I will try these when I have a minute.

felHR85, 2015-04-08:
Hey @pieman64 
I had a bug related with the CDC interface in UsbSerial. I've just updated SerialPortExample and maybe that's why it wasn't working with your Arduino clone.

pieman64, 2015-04-09:
Hi Felipe,
I just tried the new SerialPortExample with a couple of Android devices (including Samsung Galaxy Tab 2 10.1) that have root and the Arduino clones. It recognises that the USB port is connected but then states device is not supported.

felHR85, 2015-04-09:
Please, try to connect your Arduino clones with my app [DroidTerm](https://play.google.com/store/apps/details?id=com.felhr.droidterm).
It uses the same library so it is a good indicator. Also has an usb viewer to get info of what you connected. I would appreciate that info :)

pieman64, 2015-04-09:
DroidTerm also states device not supported for HL-340 USB-Serial adapter Vendor: QinHeng Electronics. Vendor Id 0x1a86 Product Id 0x7523.

felHR85, 2015-04-09:
I didn't even know that one. Probably is not supported. What kind of interfaces they use? If you use UsbViewer in Android will appear more info about the device.

pieman64, 2015-04-09:
Endpoint 1/3
Attributes 0x0002 IN
Type 0x0000 BULK
Packet Size 0x0020(32)

Endpoint 2/3
Attributes 0x0002 OUT
Type 0x0002 BULK
Packet size 0x0020(32)

Endpoint 3/3
Attributes 0x0003 IN
Type 0x0003 INT
Interval 0x0001
Packet Size 0x0008(8)

felHR85, 2015-04-09:
Any info about the interface? Maybe it is implemented as a Vendor specific interface and that would be a reason because it is not working.

pieman64, 2015-04-09:
Viewer states:
Interface(1/1)
Class: 0x00ff Vendor Specific Class.

felHR85, 2015-04-09:
That's the problem. I found some resources so I will try to develop the driver for that device but I would probably need your help for testing.

pieman64, 2015-04-09:
No problem with us testing Felipe. Just let us know when you need us.

felHR85, 2015-04-09:
I just added a new version of SerialPortExample with, I hope, support for those arduino clones. I've added a lot of log traces so in case of not working we will know where are issues.  Logs should start with the name CH34xSerialDevice

pieman64, 2015-04-09:
Felipe I can see the revised jar but do you also have a revised apk?

felHR85, 2015-04-09:
Apk updated!

pieman64, 2015-04-09:
App still states USB device not supported. I have emailed you the alogcat if it helps.

closed 2015-06-02 by felHR85

bigmachini, 2017-01-23:
How did you end up resolving this. having the same problem now...