felHR85/UsbSerial #7: Support of new devices

felHR85, 2015-04-09:
Currently UsbSerial supports FT232, PL2303, CP210x and CDC serial device. If you know any Usb to Serial converter which is not supported please, list them here. 

Felipe

bejibx, 2015-07-20:
I'm trying to communicate with [Cypress FX3 chip](http://www.cypress.com/products/ez-usb-fx3-superspeed-usb-30-peripheral-controller) using your library but with no luck for now. Don't know what kind of serial device they use though.

felHR85, 2015-07-20:
Two possible sources of the problem:
- They use a vid and pid that I didn't register.
- They use a custom solution, so a driver for that device should be developed.

The first source of problems should be easy to solve. Instead of use
~~~
UsbSerialDevice serial = UsbSerialDevice.createUsbSerialDevice(device, usbConnection); 
~~~

Try to load the each driver individually to see if someone works
~~~
CDCSerialDevice  serialDevice = new CDCSerialDevice(device, usbConnection);
CH34xSerialDevice serialDevice = new CH34xSerialDevice(device, usbConnection);
CP2102SerialDevice serialDevice = new CP2102SerialDevice(device, usbConnection);
FTDISerialDevice serialDevice = new FTDISerialDevice(device, usbConnection);
PL2303SerialDevice serialDevice = new PL2303SerialDevice(device, usbConnection);
~~~

felHR85, 2015-07-20:
I found Cypress has his own USB-UART controller. Probably they used it http://www.cypress.com/documentation/datasheets/cy7c65213-usb-uart-lp-bridge-controller

anddax, 2016-02-04:
Please add Moxa USB-Serial device VID 0x110a  PID 0x1150.   Thank You!

felHR85, 2016-02-04:
What chipset does it use? FTDI, CP210x... @anddax

anddax, 2016-02-04:
Probably Prolific Pl2303.   On Windows is show Prolific 067B VID2303. 
 On Android Your Program not found it.
On Fedora Linux is show VID 0x110a PID 0x1150 vendor MOXA.

vmsJames, 2016-02-11:
Here's a new device that Intermec is using - when their serial adaptor is attached onto the device (CN51), you can then find:
VID: 0x067e
PID: 0x2800

I asked them what chipset they're using: Texas Instruments TUSB3410

felHR85, 2016-02-13:
@anddax I've added the Moxa Uport 1150 in the UsbSerial 4.0 release

felHR85, 2016-02-13:
@vmsJames Actually UsbSerial doesnt support the TUSB3410 chipset. I will try to get a cheaper one.

ge0rg, 2016-03-09:
[FTD312D](http://www.ftdichip.com/Products/ICs/FT312D.html) seems to be a very interesting newcomer. It's a UART module that provides an USB Host to Android's [AOA](https://source.android.com/devices/accessories/protocol.html) interface and that seems to allow charging of the Android device in operation. The product page is listing a demo app and SDK documentation, but so far I failed to find the relevant downloads. Still, it looks like a very interesting choice for DIY projects.

felHR85, 2016-03-10:
Hello @ge0rg !
It seems a very interesting module. I will try to gather information and I wouldnt mind to buy a cheap one for testing.

pcancela, 2016-03-14:
Hello felHR85,

It would be great if your library supports Intermec's Usb Serial adapter p/n 225-737-002 with VID: 0x067e - PID: 0x2800. 

Is it really a Texas Instruments TUSB3410 chipset?

Is there anything I can do to help?

Cheers

felHR85, 2016-03-14:
Hello @pcancela 
I don't have any of those USB serial adapters so I don't really know what chipset they use.
The greatest help would be letting me get one of them as I am always open to add support for new chipsets ;)

Leftyist, 2016-04-07:
Hey felHR85,

Any chance of supporting chip CP2130 in the future? I'm trying to implement it now, but having some difficulties. Otherwise, would you have any tips to offer?

Thanks!

felHR85, 2016-04-07:
Hi @Leftyist 
I didn't  know that one. I am always open to support new devices. I would do the implementation if I could got one.
[Here it is the documentation, it looks pretty good](https://www.silabs.com/Support%20Documents/TechnicalDocs/AN792.pdf)

felHR85, 2016-05-01:
I just added support for CP2130 and its SPI interface. Hope you can take a look @Leftyist

echo22, 2016-06-22:
Hi, I'm trying to talk to a STMicroelectronics Nucleo F401RE but for now without success. On the log console I have this message I/CDCSerialDevice: Control Transfer Response: -1. It does not happens if I use an Arduino, wich works fine. What kind of problem could be?
Thanks! Nice library!

lucianommatos, 2016-06-24:
Hello Felipe, I'm using your library to connect an Arduino UNO R3 to my Android smartphone and trying to light up a LED from the app but I have problems... I've checked and figured out that maybe the PID and VID are not supported by this library. How would I solve this problem? Should I make a fork and solve this? Can you guide me? hahaha! Thanks! By the way, congratulations for your library!
(Picture of the toast arduino info)
[image]

red15530, 2016-08-31:
It does not work with a USB microcontroller PIC18F14K50 VID 0x0408 PID 0x000A

Chewie9999, 2017-01-25:
It doesn't work with Juniper SRX300 routers.
On the Droidterm Pro error it says "USB Device not supported: Unknown device
Vendor: Cygnal Integrated Products, Inc."
In Windows, it shows up as the following:
USB\VID_10C4&PID_8470&REV_0101
Interestingly, under windows the Manufacturer says "Silicon Laboratories", not Cygnal.
The Human-readable driver name is "Juniper Networks BX Series System Console" if you want a more generic name for it.
Would it be possible to add support for this device?

Thank you!

Omcsesz, 2017-05-01:
It doesn't work with FT311D Ic.

kovleventer, 2017-08-04:
The E+H HART Commubox uses the CP210x chipset, but I could not find the corresponding entry (VID: 0x10C4, PID: 0x8135), so I had to add that directly in my project.

mobilinkd, 2020-06-12:
Please support STM32 Nucleo boards.  This library does not work with any STM32 Nucleo board I have used.  These appear to be normal CDC devices and seem to be detected as such.  However, they have multiple interfaces (debug, mass storage and CDC), not just a CDC interface.

These work just fine with the mik3y/usb-serial-for-android library and applications which use it.  But applications which use this library detect the port and appear to attach to it, but then no data transfers occur.  I have only rudimentary knowledge of USB CDC spec.  I have created hardware which uses the STM32 CDC driver -- which does work with this library, but the way.

It seems that the way the felHR85 library and the mik3y library handle the control interface differ a bit.  mik3y determines the control index when identifying the CDC component of a device with multiple USB interfaces and passes that index when calling connection.controlTransfer().  This library always passes an index of 0 to the same call.

I have pasted the relevant output of `lsusb -v -d 0483:374b` below.  If you want me to send one of these boards to you, I am happy to do so.

```
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         2
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               6 ST-Link VCP Ctrl
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              6 ST-Link VCP Ctrl
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          3
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        2
        bSlaveInterface         3 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              7 ST-Link VCP Data
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               0
```

mobilinkd, 2020-06-13:
https://github.com/felHR85/UsbSerial/pull/314 addresses the issue with STM32 Nucleo boards.

scarycz, 2022-01-18:
Hello, any chance of adding newer Prolific PL2303G series in future? - https://prolificusa.com/product-category/chips/i-o-products/usb-to-uart-g-series/

Unrelated page, nevertheless some changes are summarised in https://reviews.freebsd.org/D27952