felHR85/UsbSerial #9: USB/Serial device doesn't show up in UsbManager#getDeviceList()

BenLeggieroNCR, 2015-11-18:
I need to use this (os a similar) library to interface with a device which is connected like this: [image]

but it doesn't show up when I call `usbManager.getDeviceList()`. I do get multiple devices, but not the one I connected like above (I know the other devices' `vendorId`s, plus the list size doesn't change when I plug/unplug the device).

felHR85, 2015-11-18:
If usbManager.getDeviceList() is getting no response it is an Android issue. I would you recommend you trying with my app DroidTerm and see if it works properly

BenLeggieroNCR, 2015-11-18:
It is getting a response. The response has two devices in it, but those are known devices, and they are the only devices in that list, regardless of whether External Device is plugged in.

I will try to use DroidTerm and get back to you.

BenLeggieroNCR, 2015-11-18:
It is getting a response. The response has two devices in it, but those are known devices, and they are the only devices in that list, regardless of whether External Device is plugged in.

I don't see how I can use DroidTerm, since this is not a device with the Google Play Store. Do you have an APK or the source code, so I can install it on our Android Device?

felHR85, 2015-11-18:
Try to test with other phones or tablets too.

BenLeggieroNCR, 2015-11-18:
We don't have phones or tablets with USB-A F ports, nor a USB-A F to USB Micro-B M cable.

BenLeggieroNCR, 2015-11-18:
That appears to be incompatible with Android API 17. Do you have an email where we can continue this?

felHR85, 2015-11-19:
felhr85 at gmail.com

closed 2015-11-28 by felHR85

BenLeggieroNCR, 2015-11-30:
This turned out to be a cable issue. It was resolved by using a cable with a PL2303 chipset.