felHR85/UsbSerial #10: Serial Read on Android API 17

neilgrewal, 2015-11-19:
I'm able to write bytes using the your library to an Arduino Mega ADK by writing a 1 or 0 to the COM port, but when trying to read bytes, the callback is never accessed. For every write to the Arduino, a response is sent back to Android but it cannot be read. I've tried using DroidTerm as well but same result. Tested and working on Android 5.1.1 but doesn't work for 4.2.2.

felHR85, 2015-11-19:
There were problems with the Android USB api below 4.3 but my code was using 4.2 as threshold. I've updated the SerialPortExample with a new UsbSerial.jar. Could you give it a try?

neilgrewal, 2015-11-24:
I had updated my USBSerial.jar earlier to the latest and that didnt have any affect. Still can't get Serial Read to work

felHR85, 2015-11-24:
Mmm I will try to add a way to force synchronous transmissions programatically as soon as possible.

BenLeggieroNCR, 2015-11-24:
If this is related, I am also having issues with separate threads issuing read and write commands asynchronously. I ended up hacking a solution together using Java's [`synchonized`](https://docs.oracle.com/javase/specs/jls/se6/html/statements.html#255769) block and [`Thread#sleep(long)`](https://developer.android.com/reference/java/lang/Thread.html#sleep(long))

felHR85, 2015-11-24:
No it shouldn't. Write operations can be performed from multiple threads

fabio978, 2016-01-11:
Hi, thank you for this library, is the only one i could make it working between Android and Arduino.
I am on Android 4.2.2 (API 17), and i have problems while receiving (length doesn't matter) strings from Arduino: sometimes they are empty or incomplete/splitted. If I use a serial monitor app on the same Android device, it always correctly read the strings.
In the past i was used to "wait" for a string terminator char before using it, because i was getting char by char. Am i doing something wrong?

Thank you very much

BenLeggieroNCR, 2016-01-12:
@fabio978 I experienced the same thing. This is because certain USB-Serial devices send the data character-by-character instead of as a whole string. I got around it by waiting 100ms after every read, to give UsbSerial time to buffer up all the data so I can receive it in one chunk.

felHR85, 2016-01-12:
Hi @fabio978 
@BenLeggieroNCR is right. Some devices can split data. You can buffer it till everything is receive, this is what I've usuallly done myself.

fabio978, 2016-01-13:
Thanks for your reply. Should i implement a buffer on my app, then?

felHR85, 2016-01-13:
I would do it in that way appending any received character to a StringBuffer till it meets your criteria.

felHR85, 2016-01-13:
In the long run I would like to add a more synchronous way to allow both ways of receiving data.

jamesjude123, 2018-10-01:
Hey brother, we have same problem, i can write but i dont have any response from what i write, is there any problem  what i code?

  ```
  public void sendCommandData(String id ,UsbSerialDevice usbSerialDevice,String data)
    {
        this.id = id;
        usbSerialDevice.write(data.getBytes());
        usbSerialDevice.read(deviceRespond);
    }

    public void openDevice(UsbDevice device, int setBaundrate, SerialCommuncationListener.OpenDeviceListener listener)
    {
        UsbSerialDevice serialDevice;
        UsbDeviceConnection connection = usbManager.openDevice(device);
        serialDevice = UsbSerialDevice.createUsbSerialDevice(device, connection);
        if (serialDevice != null) {
            if (serialDevice.open()) {
                serialDevice.setBaudRate(setBaundrate);
                serialDevice.setDataBits(UsbSerialInterface.DATA_BITS_8);
                serialDevice.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialDevice.setParity(UsbSerialInterface.PARITY_NONE);
                serialDevice.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialDevice.read(deviceRespond);
                listener.onConnectDeviceSuccess(serialDevice);
            } else {
                listener.onConnectDeviceFailure("Device can not open");
            }
        } else {
            listener.onConnectDeviceFailure("Serial is null");
        }
    }

    public void disconnect(UsbSerialDevice serialDevice)
    {
        if(serialDevice != null)
        serialDevice.close();
    }

    private UsbSerialInterface.UsbReadCallback deviceRespond = new UsbSerialInterface.UsbReadCallback() {
        @Override
        public void onReceivedData(byte[] arg0)
        {
            Toast.makeText(mContext,"RESPONSE TERMINAL: "+arg0,Toast.LENGTH_LONG ).show();
            mCallback.onDeviceResponse(id,arg0);
        }

    };
```

 UsbSerialInterface.UsbReadCallback deviceRespond no response at all

closed 2018-11-03 by felHR85

rajattechnostacks, 2023-01-25:
Yes, I want to read data from USB whenever a USB device gives a flag. are there any methods to do so.?