felHR85/UsbSerial #11: consistent read/write bytes problem

VPlatoon, 2015-11-22:
Please help me I am in confused of not be able to receiving and sending a byte! I use Android 4.3 and write an app using UsbSerial to send a byte to my Arduino with Silicon Lab CP21xx device. I am based on your example code so that everything is OK. The problem is the byte I am sending is not correctly received at arduino side. For instance if I want to send byte 0x01, then:

    cmd = HexData.stringToBytes("0x38");
    serialPort.write(cmd);

On the Arduino side, I have a very simple code as follows

    if (Serial.peek() == -1)
        return;
    char cmd = Serial.read(); // get the byte
    Serial.print(cmd); // send back to verify content
    if (cmd == 0x38)
    // ....

The callback on the Android side is:
    
    private UsbSerialInterface.UsbReadCallback callback = new UsbSerialInterface.UsbReadCallback(){
        @Override
        public void onReceivedData(byte[] args) {
                ToasterNotifier.flash(activity, "received:" + HexData.hexToString(args)); // show a toast
        }
    };

The problem is, **the byte I sent and the feedback is not identical!** For instance if I send 0x38 (le number '8'), then I get 0xF8 and then 0xFE (two separate times). The weird thing is, if I use DroidTerm (it is claimed that DroidTerm uses the library UsbSerial) to communicate with my Arduino program, it outputs correctly the sent byte and the `if` in my Arduino program is validated. 

(I set baud rate, parity, stopbits,... identical between DroidTerm and my Android app). 

Could you explain what did I miss here? Thanks!

felHR85, 2015-11-22:
Hi,
The first code you show is not correct. Try this
~~~
byte[] cmd = new byte[]{(byte) 0x38};
serialPort.write(cmd);
~~~

Don't use HexData methods there because it is not intended in that way, they are for debugging.

VPlatoon, 2015-11-27:
Hi,

I found that the connection just works when the baud rate is set to 9600. Other baud rate such as 19200 does not work. When 9600 is used, either `HexData.stringToBytes("0x38");` or `new byte[]{(byte) 0x38}` will work. But with DroidTerm any baud rate can work.

felHR85, 2015-11-27:
9600 in your Arduino code or in Android code?

felHR85, 2015-11-27:
SerialPortExample is in 9600 by default.

VPlatoon, 2015-11-27:
Of course I always set the same baud rate for Android and Arduino.

felHR85, 2015-11-28:
It would be nice to take a look of your code to see if something wrong is going on

closed 2018-11-03 by felHR85