felHR85/UsbSerial #12: NullPointerException in WorkerThread

BenLeggieroNCR, 2015-11-30:
This occurs if I do rapid start/stop operations:

    11-30 14:26:55.400 E/AndroidRuntime: FATAL EXCEPTION: Thread-242
    11-30 14:26:55.400 E/AndroidRuntime: java.lang.NullPointerException
    11-30 14:26:55.400 E/AndroidRuntime:     at com.felhr.usbserial.UsbSerialDevice$WorkerThread.run(UsbSerialDevice.java:211)

felHR85, 2015-11-30:
Some code to get the context of the issue would help :)

BenLeggieroNCR, 2015-11-30:
Here's pseudocode of my code that lead to the problem: https://drive.google.com/file/d/0B5F-QovDhUozdjVDUE1fSnJOSms/view?usp=sharing

felHR85, 2015-11-30:
Probably the easiest way is to hold the resources (I don't think it would be noticeable). UsbSerial must start some threads and that takes some time

BenLeggieroNCR, 2015-11-30:
Is there a way for me to listen for when UsbSerial has started up and is ready to go?

felHR85, 2015-11-30:
Start UsbSerial when the service (or Activity is created, as shown in SerialPortExample) should be enough. Read operation is not intended to be use every time you want listen something. I wouldn't implement the fetchDataFromDeviceOnlyOnce. Just add the code you want to use in that case inside the callback method.

BenLeggieroNCR, 2015-11-30:
So the solution is to keep my continuous-polling thread running in the background for the entire lifecycle of the app?

felHR85, 2015-11-30:
UsbSerial does not need poll anything. That callback is only called when something is read from the USB virtual serial port. Just use serialPort.read(mCallback); at the beginning of your code among the other calls like setting the baud rate, data bits...
Internally UsbSerial uses the asynchronous usb api for 4.3 and newer Android versions. Older versions use the synchronous api with the timeout parameter set to 0.
Serial ports are usually opened in that way. Open the port and keep it open. It is not going to drain the device battery, the UI will be as smoother as always because read/write ops are performed in other threads.

BenLeggieroNCR, 2015-11-30:
I see; you're suggesting I shut down the polling thread, but keep the UsbSerial object and its read callback alive. That should work.

felHR85, 2015-11-30:
I wouldn't worry about threads and UsbSerial because write operations are enqueue and transmitted when possible (you can call the write method within the UI) and read callback is executed in its own thread. SerialPortExample has a good example of this (UsbService.java), check it out

closed 2015-11-30 by BenLeggieroNCR