felHR85/UsbSerial #14: can't write after setting parameters immediately

boogerlad, 2015-12-20:
<pre>
serial.setBaudRate(115200);
serial.setDataBits(UsbSerialInterface.DATA_BITS_8);
serial.setStopBits(UsbSerialInterface.STOP_BITS_1);
serial.setParity(UsbSerialInterface.PARITY_NONE);
serial.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
//serial.write("sa".getBytes());
serial.read(mCallback);
//serial.write("sa".getBytes());
</pre>

serial.write does not work if you uncomment either of the commented lines, or both. In the demo application, it works by pressing a button. I would like to write to it as soon as it is ready. Is there a synchronous version of the functions?

BenLeggieroNCR, 2015-12-28:
+1 I'd love some sort of way to know when this library is ready. My current workaround is like yours; wait for the user to take an action that will write.

felHR85, 2015-12-28:
I've tried both ways and it works fine (At least with a CDC device but it shouldn't be any problem with other chipsets). I would put write after read by the way.
Please check out your code because I've used it the lib in that way a lot of times.
Merry xmas!

BenLeggieroNCR, 2015-12-28:
I'm using a Prolific chipset, maybe that'll throw some strangeness.

felHR85, 2015-12-28:
I've just tried with my PL2303 and it works.

felHR85, 2015-12-28:
There are Prolific clones out there. Maybe you are using one of them

BenLeggieroNCR, 2015-12-28:
@felHR85 perhaps. Still, you mentioned to wait for worker threads to spin up.

felHR85, 2015-12-28:
I just added a safeguard to avoid that (UsbSerialDevice.java lines 361 and 381).

BenLeggieroNCR, 2015-12-29:
@felHR85 Ah, you mean [these](https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/UsbSerialDevice.java#L361) [loops](https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/UsbSerialDevice.java#L381)? Seems shaky. Perhaps Java's `synchronized` blocks would handle that better.

felHR85, 2015-12-29:
Not sure, what I want there is a way to block the open() method till everything is ready (it shouldn't be a long waiting in 9x% percent of devices out there). I could add a synchronized wait method and wait till the threads notify they are correctly opened...Someone updated the project to AndroidStudio and I accept pull requests so if you can give it a try I am open to pull every change that means improvement.

BenLeggieroNCR, 2015-12-29:
@felHR85 if we need to get back to this, I'll see what I can do.

Vijayapriyacs, 2017-05-18:
Attempt to invoke virtual method 'void com.felhr.usbserial.UsbSerialDevice.write(byte[])' on a null object reference
 when i call serialport.write("DATA".getBytes); i got following error. how to resolve this

Vijayapriyacs, 2017-06-30:
Any body  help me to fix this problem.

closed 2018-11-03 by felHR85