felHR85/UsbSerial #15: unable to print results

boogerlad, 2015-12-20:
<pre>
serial.setBaudRate(115200);
serial.setDataBits(UsbSerialInterface.DATA_BITS_8);
serial.setStopBits(UsbSerialInterface.STOP_BITS_1);
serial.setParity(UsbSerialInterface.PARITY_NONE);
serial.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
serial.read(new UsbSerialInterface.UsbReadCallback()
{
    @Override
    public void onReceivedData(byte[] arg0)
    {
        String data = new String(arg0);
        System.out.println(data);
        //Toast.makeText(getContext(), data, Toast.LENGTH_SHORT);
        //Log.i("", data);
        for(byte f : arg0)
        {
            System.out.println((char)f);
        }
    }
});
</pre>
using a toast or trying to log it causes an exception. Printing it character by character works okay. But printing the whole thing as a string doesn't work at all. I have tried using utf-8 as well and a try catch, that doesn't change anything. How are you supposed to get results? I've seen `SerialPortExample` but that is far too complex to be "client" code.

felHR85, 2015-12-20:
Hi,
More information about the exception would be helpful (You can't make a toast inside that callback because it is not executed in the UI thread).

Sometimes one string can be received split (only with certain chipsets) because the chipset behavior. For example "DATA" could be received:

D
ATA

In two calls of onReceivedData. It is good to keep that in mind.
SerialPortExample is easy (and shows some good Android patterns)!!! Use it as a template.

boogerlad, 2015-12-22:
If there is no foreach loop, and only `System.out.println(new String(arg0));`, I get nothing from stdout. I will reread SerialPortExample with a clear mind. I was a little frustrated last time I attempted.

closed 2016-01-07 by boogerlad