felHR85/UsbSerial #16: Add example sample

StephaneBg, 2016-01-05:
Please add a module with an example sample. It is simpler to look at code sometimes.

BenLeggieroNCR, 2016-01-05:
Do you mean this? https://github.com/felHR85/SerialPortExample

StephaneBg, 2016-01-05:
Yes, I meant that! Thank you!
Usually, example is provided in the same repository. My mistake.

BenLeggieroNCR, 2016-01-05:
@StephaneBg  no prob!

felHR85, 2016-01-05:
Hi @StephaneBg 
Now that UsbSerial has been ported to Android Studio maybe it would be better to add a SerialPortExample module here. What do you think?

StephaneBg, 2016-01-05:
I'm working to port the example on gradle build system. Once this will be
made, it will be possible to merge it as a module in this repository.

felHR85, 2016-01-05:
Thank you! @StephaneBg

closed 2016-01-05 by felHR85

BenLeggieroNCR, 2016-01-05:
@felHR85 I disagree; I prefer to keep libraries and their example apps separate, so those who just want to fork your repo don't have to download and manage/remove excess code they may never use.

felHR85, 2016-01-05:
@BenLeggieroNCR It is a valid point. SerialPortExample are just two java files wouldnt be a great overhead in this particular scenario. Let's keep them separate then

BenLeggieroNCR, 2016-01-05:
@felHR85 2 manually-written Java files, 3 generated ones, and 112 other files, totaling 19.4 MB. IMO a considerable overhead if the developer in question doesn't need an example.

felHR85, 2016-01-05:
Other files are generated because Eclipse's fault (and mine too because there is no .gitignore added to avoid bin and other folders).

StephaneBg, 2016-01-05:
As this library is njot published on Maven, example should be in the same
project than the library for gradle integration such as:

dependencies {

compile project(':lib')

}

felHR85, 2016-01-05:
Gradle integration would be great. Important reason to add it as a module

BenLeggieroNCR, 2016-01-05:
To ignore files generated by Eclipse, use [a .gitignore specialized for Eclipse](https://github.com/github/gitignore/blob/b6f1c2f2a0daa14953ec1ed665a7063e4322837d/Global/Eclipse.gitignore)