felHR85/UsbSerial #28: Provide a default `device_filter.xml` resource

ge0rg, 2016-02-23:
The USB Host SDK provides [an intent filter for USB attach events](http://developer.android.com/guide/topics/connectivity/usb/host.html#manifest-example), that can be used to auto-launch your app when a certain USB device is attached.

It would be great to have such a `device_filter.xml` file as part of the usbserial source code (or even AAR), based on the vendor/product IDs supported by UsbSerial. It is possible to create a momentary snapshot of the supported IDs thanks to the consistent style in the `deviceids` directory, using this black magic incantation (you just need to mix in some XML boilerplate):

    perl -lne 'print "\t<usb-device vendor-id=\"$1\" product-id=\"$2\" />" if /new ConcreteDevice.0x([0-9a-f]{4}), 0x([0-9a-f]{4})/i' java/com/felhr/deviceids/*.java

However, it would be nicer and cleaner to have the XML either auto-generated in the build process or (manually?) updated with new device IDs. It is probably also possible to replace many of the entries by a vendor/class or similar combination, instead of listing all supported IDs.

felHR85, 2016-03-10:
I guess this would allow to auto open the serial port when the device is just connected. Am I right?

ge0rg, 2016-03-10:
Exactly. If you have only one supporting app installed, it is opened directly; if you have multiple apps, you get a ttypical chooser intent dialog.

csobsidian, 2016-07-19:
This should be implemented by the application and not by the library. Including it by default would mean every app would have this behavior even if they didnt want it.

ge0rg, 2016-07-29:
The application needs to include a device filter XML for this feature to work. My suggestion is merely to provide such an XML file as part of the library, so that app developers have an easier time integrating.

bill-chadwick, 2017-12-14:
Indeed XML file(s) ready populated with the vid+pid pairs in the /java/com/felhr/deviceids namespace source files would be very handy.

bill-chadwick, 2017-12-14:
<?xml version="1.0" encoding="utf-8"?>
<resources>
	
	<usb-device vendor-id="17224" product-id="21795" />
	<usb-device vendor-id="6790" product-id="29987" />
	<usb-device vendor-id="6790" product-id="21795" />
	<usb-device vendor-id="6790" product-id="1093" />

	
	<usb-device vendor-id="1115" product-id="83" />
	<usb-device vendor-id="1137" product-id="1642" />
	<usb-device vendor-id="1161" product-id="57344" />
	<usb-device vendor-id="1161" product-id="57347" />
	<usb-device vendor-id="1861" product-id="4096" />
	<usb-device vendor-id="2118" product-id="4352" />
	<usb-device vendor-id="2278" product-id="21761" />
	<usb-device vendor-id="2301" product-id="10" />
	<usb-device vendor-id="3053" product-id="4352" />
	<usb-device vendor-id="3053" product-id="4353" />
	<usb-device vendor-id="4047" product-id="4099" />
	<usb-device vendor-id="4047" product-id="4100" />
	<usb-device vendor-id="4047" product-id="4102" />
	<usb-device vendor-id="4062" product-id="51717" />
	<usb-device vendor-id="4262" product-id="43558" />
	<usb-device vendor-id="4267" product-id="4293" />
	<usb-device vendor-id="4277" product-id="44144" />
	<usb-device vendor-id="4292" product-id="3985" />
	<usb-device vendor-id="4292" product-id="4353" />
	<usb-device vendor-id="4292" product-id="5633" />
	<usb-device vendor-id="4292" product-id="32778" />
	<usb-device vendor-id="4292" product-id="32827" />
	<usb-device vendor-id="4292" product-id="32836" />
	<usb-device vendor-id="4292" product-id="32846" />
	<usb-device vendor-id="4292" product-id="32851" />
	<usb-device vendor-id="4292" product-id="32852" />
	<usb-device vendor-id="4292" product-id="32870" />
	<usb-device vendor-id="4292" product-id="32879" />
	<usb-device vendor-id="4292" product-id="32890" />
	<usb-device vendor-id="4292" product-id="32964" />
	<usb-device vendor-id="4292" product-id="32970" />
	<usb-device vendor-id="4292" product-id="32989" />
	<usb-device vendor-id="4292" product-id="33014" />
	<usb-device vendor-id="4292" product-id="33045" />
	<usb-device vendor-id="4292" product-id="33085" />
	<usb-device vendor-id="4292" product-id="33087" />
	<usb-device vendor-id="4292" product-id="33098" />
	<usb-device vendor-id="4292" product-id="33099" />
	<usb-device vendor-id="9221" product-id="3" />
	<usb-device vendor-id="4292" product-id="33110" />
	<usb-device vendor-id="4292" product-id="33118" />
	<usb-device vendor-id="4292" product-id="33119" />
	<usb-device vendor-id="4292" product-id="33163" />
	<usb-device vendor-id="4292" product-id="33183" />
	<usb-device vendor-id="4292" product-id="33190" />
	<usb-device vendor-id="4292" product-id="33193" />
	<usb-device vendor-id="4292" product-id="33196" />
	<usb-device vendor-id="4292" product-id="33197" />
	<usb-device vendor-id="4292" product-id="33224" />
	<usb-device vendor-id="4292" product-id="33250" />
	<usb-device vendor-id="4292" product-id="33255" />
	<usb-device vendor-id="4292" product-id="33256" />
	<usb-device vendor-id="4292" product-id="33266" />
	<usb-device vendor-id="4292" product-id="33304" />
	<usb-device vendor-id="4292" product-id="33323" />
	<usb-device vendor-id="4292" product-id="33387" />
	<usb-device vendor-id="4292" product-id="33409" />
	<usb-device vendor-id="4292" product-id="33427" />
	<usb-device vendor-id="4292" product-id="33529" />
	<usb-device vendor-id="4292" product-id="33601" />
	<usb-device vendor-id="4292" product-id="33666" />
[... 656 lines cut ...]
	<usb-device vendor-id="1841" product-id="8195" />
	<usb-device vendor-id="1293" product-id="599" />
	<usb-device vendor-id="1423" product-id="38688" />
	<usb-device vendor-id="4598" product-id="8193" />
	<usb-device vendor-id="1962" product-id="42" />
	<usb-device vendor-id="1453" product-id="4026" />
	<usb-device vendor-id="21362" product-id="8963" />
	<usb-device vendor-id="1008" product-id="2873" />
	<usb-device vendor-id="1008" product-id="12601" />
	<usb-device vendor-id="1008" product-id="12857" />
	<usb-device vendor-id="1008" product-id="13604" />
	<usb-device vendor-id="1208" product-id="1313" />
	<usb-device vendor-id="1208" product-id="1314" />
	<usb-device vendor-id="1356" product-id="1079" />
	<usb-device vendor-id="4525" product-id="1" />
	<usb-device vendor-id="2915" product-id="25904" />
	<usb-device vendor-id="2956" product-id="8963" />
	<usb-device vendor-id="4362" product-id="4432" />
	<usb-device vendor-id="1367" product-id="8200" />

	
	<usb-device vendor-id="9805" product-id="562" />
	<usb-device vendor-id="9805" product-id="288" />

</resources>

bill-chadwick, 2017-12-14:
[device_filter.txt](https://github.com/felHR85/UsbSerial/files/1559569/device_filter.txt)