felHR85/UsbSerial #33: i have error Error: Failed to resolve: com.github.felHR85:UsbSerial:4.1.1

XSparter, 2016-03-07:
IS possible to resolve it?

felHR85, 2016-03-07:
Let me take a look to see what is happening @XSparter

XSparter, 2016-03-07:
Sorry, I mixed the instructions . Now it works , but I have another problem .
I have this code:
`     UsbDevice device = null;
        UsbDeviceConnection usbConnection = null;
        UsbSerialDevice serial = UsbSerialDevice.createUsbSerialDevice(device, usbConnection);
        serial.open();
        serial.setBaudRate(9600);
        serial.setDataBits(UsbSerialInterface.DATA_BITS_8);
        serial.setParity(UsbSerialInterface.PARITY_ODD);
        serial.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
        serial.write(("ciao").getBytes());
        serial.close();`

But give me null pointer exception,i think because i don't have specified a device to comunicate. Can you help me to implement this function?

felHR85, 2016-03-07:
[You should check out the example](https://github.com/felHR85/UsbSerial/tree/master/example/src/main/java/com/felhr/serialportexample) and use it as a template

XSparter, 2016-03-08:
it's very hard for my need. I need simply code for write in serial... Do you have a small serial write and read example?

BenLeggieroNCR, 2016-03-08:
@XSparter That's what that is. See [line 84 in MainActivity](https://github.com/felHR85/UsbSerial/blob/master/example/src/main/java/com/felhr/serialportexample/MainActivity.java#L84) and [line 249 in UsbSerial](https://github.com/felHR85/UsbSerial/blob/master/example/src/main/java/com/felhr/serialportexample/UsbService.java#L249).

XSparter, 2016-03-08:
i have resolved with this.
https://github.com/mik3y/usb-serial-for-android/releases
Sry,but this library dosen't worked for me...

closed 2016-04-15 by felHR85

hafidben, 2017-04-05:
you can resolve it : https://github.com/felHR85/UsbSerial/issues/54