felHR85/UsbSerial #36: Problem converting received data to float

smith1994, 2016-03-21:
I want to get the received data and add to my chart, but I can't convert the received data to float.
Here's my code:
```
public void onReceivedData( byte[] arg0) {
            String data_tv = null;
            try {
                data_tv = new String (arg0,"UTF-8");
                data=data_tv;
                data_tv.concat("/n");
                tvAppend(textView, data_tv);
                float fdata = Float.parseFloat(data_tv);

            } catch (UnsupportedEncodingException e) {
                e.printStackTrace();
            }       
        }
```
 I've tried this, but it doesn't work.
`float f = ByteBuffer.wrap(arg0).order(ByteOrder.LITTLE_ENDIAN).getFloat();`
Please help!!!!!

felHR85, 2016-03-21:
Hi,
What's exactly happening?
Is that "tvAppend" appending text to a textview? Because that operation must be performed only in the UI thread and that code is not executed on the ui thread

smith1994, 2016-03-21:
Hi felHR85,
I want to get the sensor value from the arduino, then draw a line chart in android. You're right, "tvAppend" appending text in a TextView. 
Here's my arduino code:
```
void setup() {
  Serial.begin(9600);
}
void loop() {
  if(Serial.available())
  {
    int sensorValue = analogRead(A0);
  float voltage = sensorValue * (5.0 / 1023.0);
  Serial.println(voltage);
  delay(500);
  }
}
```
and android code:
[package-com.docx](https://github.com/felHR85/UsbSerial/files/182504/package-com.docx)

felHR85, 2016-03-21:
Don't call tvAppend from there because it is going to crash. [Use a Handler as in the example , line 64](https://github.com/felHR85/UsbSerial/blob/master/example/src/main/java/com/felhr/serialportexample/UsbService.java) 

if (mHandler != null)
                    mHandler.obtainMessage(MESSAGE_FROM_SERIAL_PORT, data).sendToTarget();

The Handler is defined on the MainActivity and is executed on the UI

smith1994, 2016-03-22:
Thanks felHR85!!! I use your example code here, but it crashed again when I tried to convert. How can I convert this data to float?
```
/*
     * This handler will be passed to UsbService. Data received from serial port is displayed through this handler
     */
    private static class MyHandler extends Handler {
        private final WeakReference<MainActivity> mActivity;

        public MyHandler(MainActivity activity) {
            mActivity = new WeakReference<>(activity);
        }

        @Override
        public void handleMessage(Message msg) {
            switch (msg.what) {
                case UsbService.MESSAGE_FROM_SERIAL_PORT:
                    String data = (String) msg.obj;
                    float fdata = Float.parseFloat(data_tv); //I added this
                    mActivity.get().display.append(data);
                    break;
                case UsbService.CTS_CHANGE:
                    Toast.makeText(mActivity.get(), "CTS_CHANGE",Toast.LENGTH_LONG).show();
                    break;
                case UsbService.DSR_CHANGE:
                    Toast.makeText(mActivity.get(), "DSR_CHANGE",Toast.LENGTH_LONG).show();
                    break;
            }
        }
    }
```

felHR85, 2016-03-22:
I would need the error trace :)

smith1994, 2016-03-22:
Here it is but instead of converting once it does tiwce. So i have 2 values is the same. 
```
try {
       fdata = Float.valueOf(data);
 } catch (NumberFormatException e) {}
```

felHR85, 2016-03-22:
So, Are yoy getting a NumberFormatException error?

smith1994, 2016-03-22:
When i don't use a try catch my application crash, so i think it is NumberFormatException error, and when i add try catch i got 2 values at the same time, the value is correct but i has two. :(

smith1994, 2016-03-22:
As you can see this is my trouble: 
[image]

felHR85, 2016-03-22:
As I understand. You want to parse the "6.0" string to 6.0 float. The NumberFormatException happens because the string is not parseable into float. One reason could be it is not guaranteed to receive 6.0 at the same time.

You could receive 
6
.
0

Or

6.
0

This happens with some devices. In that case I would create a StringBuffer to append the string and only convert it to float when is safe to do it.

Maybe this solves the problem of receiving the same value repeated but that's out of my reach i am afraid of :S

smith1994, 2016-03-22:
:) Thank you very much, i'll try this.

BenLeggieroNCR, 2016-03-23:
Parsing numbers from serial data is difficult. I solved this by throwing everything into a string buffer queue with a limited length, and then checking it every time against a regex like `/\d+(\.\d+)?/`. Then, only if it matches, I take group `$0` and parse that as a `double` and clear the buffer.

closed 2018-12-07 by felHR85