felHR85/UsbSerial #38: How to read data using

karan221291, 2016-04-05:


felHR85, 2016-04-05:
Hi @karan221291 
Some devices doesnt send the data continously (Maybe it is a problem of some Android tablets).
It is easy to solve using a buffer and appending the data there till you receive everything you want.

You could use the synchronous version (described in the README file) to set the max length you want to read but this wouldnt solve the hardware problematic.

karan221291, 2016-04-06:
@felHR85 what condition do i need to check that tells me data received from my serial machine has ended? is there any way by which i can use library for below scenario

-open connection with serial device
-get data continuously from serial device until connection is closed.

felHR85, 2016-04-06:
If you are using some kind of protocol you should know when end of data is reached.
If there is no end. Just append characters to a buffer and set the buffer to a TextView for example.

Both of them can be done easily.

karan221291, 2016-04-07:
@felHR85 thanks for quick reply, can you provide any example for how to use buffer in this situation, I don't know how to use as we will  not be able to get input & output streams.

closed 2016-05-26 by karan221291

Ayyan-coditas, 2017-01-03:
@karan221291  Did you find a solution for this? I am also stuck with this problem and it would be wise to post a solution for future readers.

mahermeg17, 2017-11-29:
Hi,
I get this problem when reading from UART port with a PICO PI IMX6UL board.
But when using an other usb library, it can read the message in one piece. So I thing it's an issue with the UsbSerial library. Sorry I don't know how to fix it.
The lib that I use to confirm : https://github.com/mik3y/usb-serial-for-android

Bests,