felHR85/UsbSerial #40: NullPointerException in WorkerThread.run

MarkoKatic, 2016-04-11:
Greetings,

We are using this library in our application to talk  with various usb serial devices. We have encountered rare instances where our service would crash with the following stack trace:

`Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.nio.ByteBuffer.isDirect()' on a null object reference
       at android.hardware.usb.UsbRequest.dequeue(UsbRequest.java:156)
       at android.hardware.usb.UsbDeviceConnection.requestWait(UsbDeviceConnection.java:230)
       at com.felhr.usbserial.UsbSerialDevice$WorkerThread.run(UsbSerialDevice.java:213)`

Our service would then restart and then works just fine.
Apparently this isn't something new, other people have had similar problems. It seems that this is a problem in the aosp usb subsystem:
https://code.google.com/p/android/issues/detail?id=185219

We initialize and read from the UsbSerialDevice like this:

        try {
            usd.open();
            usd.setBaudRate(115200);
            usd.setDataBits(UsbSerialInterface.DATA_BITS_8);
            usd.setParity(UsbSerialInterface.PARITY_ODD);
            usd.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);

            UsbSerialInterface.UsbReadCallback ucb = new UsbSerialInterface.UsbReadCallback() {

                @Override
                public void onReceivedData(byte[] data) {
                //Some code
                }
            };
            usd.read(ucb);
        } catch (Exception e) {
            LOG.error("while initializing usd {}", e);
            usd.close();
        }

And when we need to write data we just call write(byte[] buffer)

Any ideas on how to fix this? I was thinking of just wrapping the call to requestWait() in WorkerThread to catch the NPE and ignore it.

felHR85, 2016-04-11:
Hello @MarkoKatic 
I've never found this one before but it seems pretty random apparently. Does it happen to you with Android 5.x or superior?

I am not sure if this particular bug could be just ignored. In the link you provide there is a workaround that looks good but I need to check it out more deeply.

MarkoKatic, 2016-04-11:
Our hardware runs 5.0.2.

felHR85, 2016-04-17:
I just implemented a workaround I got from the Android issue tracker. Please give it a try. It is actually in a branch called bug_40.

tysm-msab, 2018-11-19:
The bug_40 branch fixed this issue for me. Still seen on Android 7/8. Any plans to merge it to main? Thanks for the fantastic code and updates!

felHR85, 2018-11-19:
@tysm-msab Really good news! I will take a deeper look this week and I will try to merge it during this week.

commit 1d78e71c (2016-04-17): SafeUsbRequest class added hoping to fix #40 bug
closed 2018-12-25 by felHR85