felHR85/UsbSerial #41: How to list the device using USBSerial?

anhdenday, 2016-04-11:
Dear @felHR85 ,
I want to list all of usb device that plug in my android device (example my android device has 3 usbs port). It's possible if I want to list all if device that attached to my android.
Please take a look at it. 
Thanks

felHR85, 2016-04-12:
```java
  private void findSerialPortDevice() {
        // This snippet will try to open the first encountered usb device connected, excluding usb root hubs
        HashMap<String, UsbDevice> usbDevices = usbManager.getDeviceList();
        if (!usbDevices.isEmpty()) {
            boolean keep = true;
            for (Map.Entry<String, UsbDevice> entry : usbDevices.entrySet()) {
                device = entry.getValue();
                int deviceVID = device.getVendorId();
                int devicePID = device.getProductId();

                if (deviceVID != 0x1d6b && (devicePID != 0x0001 || devicePID != 0x0002 || devicePID != 0x0003)) {
                    // There is a device connected to our Android device. Try to open it as a Serial Port.
                    requestUserPermission();
                    keep = false;
                } else {
                    connection = null;
                    device = null;
                }

                if (!keep)
                    break;
            }
            if (!keep) {
                // There is no USB devices connected (but usb host were listed). Send an intent to MainActivity.
                Intent intent = new Intent(ACTION_NO_USB);
                sendBroadcast(intent);
            }
        } else {
            // There is no USB devices connected. Send an intent to MainActivity
            Intent intent = new Intent(ACTION_NO_USB);
            sendBroadcast(intent);
        }
    }
```
This method is looking for this first device to meets a given criteria but every device connected is available through

 ```java
HashMap<String, UsbDevice> usbDevices = usbManager.getDeviceList();
```

anhdenday, 2016-04-14:
Thanks you, it's worked.

closed 2016-04-14 by anhdenday