felHR85/UsbSerial #45: Issue reading data from the serial port

djadry, 2016-04-28:
Hello,
I'm having some difficulties using the UsbSerial library: I'm developing an application that runs on Android 4.0.4 (so I'm using the synced methods) and I'm trying to read some bytes from the serial port. I made some testing and it resulted that the syncRead() method only reads bytes one by one. And if I try to send more bytes simultaneously, it loses them; it seems that the buffer is not properly working.

Is there some documentation to refer to? Or do you have any advice?

Many thanks, Francesco

felHR85, 2016-04-28:
Hi @djadry 
Sync methods don't use any buffer so the issue is not there. What USB to serial converter are you using?

djadry, 2016-04-29:
So, just to make it clear: do you confirm me the that the syncRead() (that I saw internally uses bulkTransfer) can only return data that arrives immediately, and so does not have an internal buffer for incoming data still not read?
If this is the case, I will have to use the async methods obviously and buffer them by myself.

I am using both Prolific (that returns no more than 1 or 2 bytes, and loses the remaining ones) and FTDI (that only returns 1 byte, but with the wrong value...).
I am using an high timeout for the moment (something like 3 seconds).

Just to explain the situation better, my goal is to use your UsbSerial as the low level communication layer for the j2mod Modbus Library, that current on Android uses jSerialComm that relays on kernel support (/dev/ttyUSBx), so it unusable for me. 
But I have to use blocking read functions with a timeout for this so I tried with the syncRead(), but I think I will have to go with async one and buffer myself...

Thanks
Bye

felHR85, 2016-04-29:
SyncRead is just calling bulktransfer method (except in FTDI case where it is filtering the 2 byte data sent every 40 ms with the line status).
I would use the asynchronous version. Check out the example module where there is a good example of it.
That project looks good. If you need some help just ask.

kubaraczkowski, 2018-09-01:
Hi,
Did either of you figure out the cause of only reading a single byte in syncRead ?
I'm getting a similar issue, though i'm sure the device (generic CDC, an mbed board)  sends e.g. 3 bytes a call to syncRead only gets 1. All following bytes are OK - they get read with the following calls to syncRead.
At the same time the timeout is not being respected - the single byte returns immediately.

Any clues?

closed 2019-07-11 by felHR85