felHR85/UsbSerial #47: How can I connect more than one usb serial devices at the same time

karan221291, 2016-05-26:
Hello felHR85 I used your library to communicate with usb serial devices and its working as i want it to.

Now requirement is little tricky here as I want to communicate using more than one usb serial devices at same time. but i am failing to do this.

Issue happening with me is that if I connect first device via usb hub to android tablet, I am able to send and receive data . now I connect another device to the usb hub and the first device is also attached. what is happening that I am only able to get the values from second device only.

I need to get values from both machines simultaneously.

karan221291, 2016-05-27:
I created multiple instances of usb interface and made them work.

karan221291, 2016-05-27:
I am able to create two threads and used them to create two instances of Usb serial, it works with devices having different kind of drivers, say CDC and PL2303. but if i connect two similar devices it is failing to communicate with them.

felHR85, 2016-06-01:
Hi,
It shouldnt be any problems connecting similar devices if it works with different ones.

karan221291, 2016-06-06:
Thanx felHR85 for your reply...I was getting incorrect ids of connected devices. so once managed them. it works fine now.

closed 2016-06-06 by karan221291