felHR85/UsbSerial #52: The app crashes if I attach a usb device, then deny the permissions and detach the usb device

josejuansanchez, 2016-06-19:
You can replicate the crash following the next steps:

1. Attach the usb device
2. Deny the permissions
3. Detach the usb device

bmdalton, 2016-06-28:
Update: if you have the issue of the app crashing after you detach the USB device (when it was functionally working at whatever you need it to do), this does NOT fix that.

felHR85, 2016-06-28:
Fix issue is still not merged. @bmdalton Did you try it on the pull request branch?

bmdalton, 2016-06-28:
I have.  I have a copy of the code running in Android Studio, so I physically made the changes there.  In both the original and the updated version, I have the same crash, which I have discovered is the result of modification of one line of code.

In your sample app, you have
[image]
while I have updated this to be
[image]

As you can see, I change  "mActivity.get().display.append(data);" to "mActivity.get().setData(dataS);" which subsequently causes the crash.  When I comment out that line, the app does not crash when I disconnect the USB device however it also doesn't function.

josejuansanchez, 2016-06-28:
@bmdalton You are right, the proposed PR in #53 only solves the issue described in the title :)

abhishekkumar1902, 2017-11-19:
any solutions for this crash issue?

closed 2019-03-09 by felHR85