felHR85/UsbSerial #57: Non sense test for excluding usb root hubs

rpelissier, 2016-07-03:
In UsbService line 182
```java
if (deviceVID != 0x1d6b && (devicePID != 0x0001 || devicePID != 0x0002 || devicePID != 0x0003)) {
                    // There is a device connected to our Android device. Try to open it as a Serial Port.
                    requestUserPermission();
                    keep = false;
                }
```

The second term of the && clause `(devicePID != 0x0001 || devicePID != 0x0002 || devicePID != 0x0003)` is always true.

I propose this fix
```java
if (deviceVID != 0x1d6b && (devicePID != 0x0001 && devicePID && 0x0002 && devicePID && 0x0003))
```

Also can you document what this codes means ?

Thank you

felHR85, 2016-07-03:
Hi @rpelissier 
You are right. Your proposal made sense.
This is done because some Android devices list the USB hubs alongside connected devices when calling getDeviceList() (even when Android docs say this call only should list connected devices)

Thank you

rpelissier, 2016-07-03:
Good to know. For more readability I propose this final fix (if 0 is also a PID we can exclude).
Thank you for sharing your great work with the community.

```java
private void findSerialPortDevice() {
        // This snippet will try to open the first encountered usb device connected, excluding usb root hubs
        HashMap<String, UsbDevice> usbDevices = usbManager.getDeviceList();
        if (!usbDevices.isEmpty()) {
            boolean keep = true;
            for (Map.Entry<String, UsbDevice> entry : usbDevices.entrySet()) {
                device = entry.getValue();

                if (isUsbRootHub(device)) {
                    // There is a device connected to our Android device. Try to open it as a Serial Port.
                    requestUserPermission();
                    keep = false;
                } else {
                    connection = null;
                    device = null;
                }

                if (!keep)
                    break;
            }
            if (!keep) {
                // There is no USB devices connected (but usb host were listed). Send an intent to MainActivity.
                Intent intent = new Intent(ACTION_NO_USB);
                sendBroadcast(intent);
            }
        } else {
            // There is no USB devices connected. Send an intent to MainActivity
            Intent intent = new Intent(ACTION_NO_USB);
            sendBroadcast(intent);
        }
    }

    private boolean isUsbRootHub(UsbDevice usbDevice) {
        return usbDevice.getVendorId() == 0x1d6b || usbDevice.getProductId() < 4;
    }
```

felHR85, 2016-07-03:
Thank you for your suggestions!! @rpelissier :)
Wouldn't be this way?
`return usbDevice.getVendorId() == 0x1d6b && usbDevice.getProductId() < 4; `

closed 2018-11-03 by felHR85