felHR85/UsbSerial #58: UsbSerialDevice Null Pointer

eXsolacyst, 2016-07-05:
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.nio.ByteBuffer.isDirect()' on a null object reference
                                                                      at android.hardware.usb.UsbRequest.dequeue(UsbRequest.java:156)
                                                                      at android.hardware.usb.UsbDeviceConnection.requestWait(UsbDeviceConnection.java:230)
                                                                      at com.felhr.usbserial.UsbSerialDevice$WorkerThread.run(UsbSerialDevice.java:213)

Line 213:
 UsbRequest request = UsbSerialDevice.this.connection.requestWait();

May need to null check UsbSerialDevice.this.connection != null before requestWait() is called.

This happens every once in a while if a UsbSerial device (Arduino in my case) is detached while running. I'm using the UsbSerialDevice.UsbReadCallback for reading so I could work around this issue by directly reading the interface but the callback is more convenient for my event driven applicaiton.

felHR85, 2016-07-05:
Hi @eXsolacyst 
This looks related to #40 
I coded a solution in a branch called bug_40
Could you give it a try?

eXsolacyst, 2016-07-05:
Will give it a try as soon as I can and let you know how it goes. Apologies
for opening a new issue for something that's already been brought to your
attention. Thanks for the awesome library btw. I really appreciate your
work.

Ryan L