felHR85/UsbSerial #59: Dont work on android 4.4 x86

akefa, 2016-07-12:
Defines but not receiving data from arduino nano, but it works fine on android 4.4 nexus 7.
What to do with android 4.4 x86? Thanks!

felHR85, 2016-07-13:
Hi @akefa 
I've not tried with Android x86 but I am not using any component in C/C++ so it should work just fine.
Maybe your Android device x86 doesnt support USB host

akefa, 2016-07-13:
USB host supported, because the camera, elm and gps-usb is connected. 
may yet what that reason might be?

felHR85, 2016-07-16:
I am not sure. 

Are you getting exceptions and crashes?
Coud you download my app DroidTerm (free version) from google play and give it a try? It would be a good way to know what's happening

dolcott, 2016-07-19:
Hi Felipe,

Not sure if this the same OS you're referring to, but DroidTerm works great on a Galaxy S4 Android 4.4.4.  I'm using it to connect to the on-chip usb on an STM32F429 processor configured as a CDC device.

One thing I see with the app is that if I open it without selecting "virtual com port" and try to close the app, the back button does nothing.  I have to force stop the app.  If I select virtual com port driver and continue, the app works great.

One question, what is the app doing when I select virtual com port, then connect?  I'm trying to do something similar using your library.  I'm following the example app in the UsbSerial repo.

I'm a bit of a newbie with android.  Thanks and great job.

Dana

akefa, 2016-07-21:
Determined that conflicts with GPS, if connect gps device disconnected, arduino detected but DroidTerm write - no supported devise VIP 046d PID 0825.

akefa, 2016-07-21:
It is possible to specify a different virtual com port?

felHR85, 2016-07-23:
@dolcott 
When you press connect Droidterm is sending configurations packets to the USB device you connected to try use it as a serial port. [Please follow thisexample](https://github.com/felHR85/UsbSerial/tree/master/example)

felHR85, 2016-07-23:
@akefa 
What GPS? Internal or external?
It is possible to open more usb virtual serial ports if you connect a USB hub to your android phone

akefa, 2016-07-24:
I use the computer (android 4.4 x86), to him connected external GPS and Arduino.
Worked with the Arduino Uno, but not nano, i have not yet determined why.

dolcott, 2016-07-24:
Hi Akefa,

Regarding not working on the nano, I found this post that might fix the problem (hardware issue)

http://forum.arduino.cc/index.php?topic=57736.0;wap2

Regards,

Dana

RDamman, 2016-10-25:
Maybe _little endian_ / _big endian_ problem in the code?