felHR85/UsbSerial #62: BBC micro:bit - detected but not working

dwhinham, 2016-07-19:
Hi there,

I'm trying to get an Android phone to talk to a BBC micro:bit, which is similar to an ARM mbed. It presents a composite device to the host; a Mass Storage class for loading firmware, a HID class, and a Communications class with CDC Interrupt endpoint and a CDC Data endpoint. The micro:bit is recognised by Mac OS X and Linux without additional drivers and can be used with terminal software.

The micro:bit uses the same USB VID/PID as the NXP LPC1768 - 0d28:0204.
However, the micro:bit uses a nRF51822 as its main processor and an Kinetis KL26 as a USB controller.

Output of `lsusb -vvv`:
```
Bus 003 Device 076: ID 0d28:0204 NXP LPC1768
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0d28 NXP
  idProduct          0x0204 LPC1768
  bcdDevice           10.00
  iManufacturer           1 MBED
  iProduct                2 MBED CMSIS-DAP
  iSerial                 3 9900023431864e45002e10060000005000000000cc4d28bd
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          122
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              7 USB_MSC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              6 MBED CMSIS-DAP
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              4 USB_CDC
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          2
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        1
        bSlaveInterface         2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              32
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              5 USB_CDC1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```
I have purchased DroidTerm PRO, but haven't tried writing my own Java code yet, so I haven't got any additional debug logs yet.

On the micro:bit I have a short program that repeatedly sends a string to the USB serial output at 115200 baud every second. DroidTerm PRO recognises that there is an "LPC1768" attached, but no text appears. It works fine on a Linux PC or Mac.

I have another program that scrolls characters across the screen of the micro:bit when they are received from the computer over serial. Again, this works on the desktop, but not with DroidTerm PRO. However, the LED on the micro:bit blinks when I send a character from DroidTerm - so *something* is happening! :)

Do you have any ideas what might be wrong, or some hints for debugging this? 
Thanks!

felHR85, 2016-07-23:
Hi @dwhinham 
Could you post a screenshot from DroidTerm USB viewer? It would be help to see how DroidTerm is seeing this hardware

dwhinham, 2016-07-26:
Hi @felHR85,

Did some more testing and found that it's actually a problem switching baud rates, so I've updated the issue title.

If I set the micro:bit firmware to use 9600 baud and DroidTerm to match, it works perfectly and I can send and receive data just fine.

If I change to any other baud rate in the firmware, it stops working (but works fine using a PC connection as already described).

I wrote a basic Android app using the library and watched the logs.
After `serial.open()` is called on the device:
```
I/CDCSerialDevice: Interface succesfully claimed
I/CDCSerialDevice: Control Transfer Response: -1
I/CDCSerialDevice: Control Transfer Response: -1
```

And then after `serial.setBaudRate()` is called:
```
I/CDCSerialDevice: Control Transfer Response: -1
I/CDCSerialDevice: Control Transfer Response: -1
```

...and the same for the `setParity()` and `setFlowControl()` functions.

Here are some screenshots of how DroidTerm Pro sees the device:
[image]
[image]