felHR85/UsbSerial #63: Could it be possible create a Processing version of this library?

q2dg, 2016-07-26:
"À la https://github.com/inventit/processing-android-serial" but working.
Thanks!

felHR85, 2016-07-27:
Hi @q2dg 
I dont know much about processing development for Android but if there is a way of using .aar libraries or interfacing java code it would be possible.

It would be good addition for UsbSerial but I would need help.

q2dg, 2016-07-27:
Well...all the information I can give you is this:

https://github.com/processing/processing/wiki/Library-Overview
https://github.com/processing/processing/wiki/Library-Guidelines
https://github.com/processing/processing/wiki/Library-Basics

It doesn't seem very difficult although my technical skills aren't good.to verify if I'm wrong
Thanks a lot