felHR85/UsbSerial #64: Wrappers for UsbSerial

felHR85, 2016-07-27:
I think the next step of UsbSerial would be adding wrappers for the most important Android frameworks
- Xamarin
- PhoneGap
- Jquery Mobile

There are more but these are probably the more important ones.

Felipe

weixingsun, 2016-07-27:
Hey,  please include react native implementation of mine using your great lib, if you like:
https://github.com/weixingsun/react-native-usbserial
tks

felHR85, 2016-07-27:
Cool! @weixingsun 
I would like to add your code to this repo. to keep all wrappers around the original lib.

weixingsun, 2016-07-27:
Alright, do you have plan to implement iOS version, too?

felHR85, 2016-07-27:
No, as far as I know USB host is not allowed on iphone/ipads

weixingsun, 2016-07-31:
@felHR85 sorry for late response, I've implemented basic functions working, like: open,read,write,close, and tested with Silicon Lab cp2102. 
Also, have you tried some chips like Silicon Lab cp2614 (I haven't yet), it ships with Apple MFi available,