felHR85/UsbSerial #68: AndroidStudio stops recognizing device after disconnect from Arduino.

aberope, 2016-08-28:
Hello, I've coded the example of this project inside my program and works excellent on CM13.1 on a Galaxy S3, but on a more new device BQ Aquaris E6, the AndroidStudio is not recognizing the Smartphone after disconnect the cable.

So the process I do is:

1. install program with studio
2. stop app from stduio (the app is uploaded)
3. connect to arduino by OTG cable, the app automatically opens.
4. the program is working fine. But studio is not reconizing the device until reboot.

felHR85, 2016-08-28:
Hola! @aberope 
My personal experience tells me that BQ devices are prone to have hardware issues. If it is working with your S3 (Cyanogen? Standard S3 USB host mode was a pain sometimes) I am afraid it's an issue with that BQ phone.

aberope, 2016-08-28:
Hola! 👍 Thanks for the fast answer and it makes me happy to know is a known issue. I have no idea of how your driver works but I need it a lot and it would be nice to find a solution. I imagine you are using an intermediate layer that vendor brings to all, and I guess it can't be fixed without root and modifying that driver from phone. Have you got any solution of this kind? I think Cyanogen is perfect if works well. The first snapshot I tried on S3 was really bad, the v8. But 13.1 is working like a charm! About communicating with serial driver, seems stable. 

I guess everything resides on the chip inside the OTG cable and the driver on the phone. You just make the connection. So, nothing good can be done without opening those things...

Thanks for your support, it's an awesome project and you can't imagine how happy made me when found it. Just I hope someday to be fully supported on any Android operating system.

aberope, 2016-08-29:
I am thinking and seems some buggy variable. When I connect to studio it won't work always, just sometimes, and reboot fixes always, but if I reconnect to Arduino the App detects the device and continues working. So it's not bugged, just some variable is not sanitized after disconnect, seems like straight away to Arduino, and avoids any other thing until this variable is sanitized by the phone's driver of your software in some way. So the problem begins when I connect it to Arduino, then will be fine to understand how the auto-allowing by XML works because maybe has relation with the bug. Due that phone only works with Arduino by USB. Can't be sent some reset to the driver from the App when the cable is disconnected?

Thanks in advance, have a good day.

felHR85, 2016-09-02:
UsbSerial uses the usb api android offers you (it is basically a wrapper over C/C++ libusb library). Api exposes the lowest level USB calls so I basically build a user-space driver using those calls.
Rooting doesnt affect UsbSerial. If you root your phone you could write C/C++ to use the usb kernel drivers but I am not sure if that would solve the BQ issue (and I know it will be a pain).

closed 2018-11-03 by felHR85