felHR85/UsbSerial #70: An issue when writing data right after syncOpen a connection

utkukaratas, 2016-08-31:
Just in case you're trying to send a one off message to the Arduino at the start of the connection, you have to sleep/delay some time in order to let the Arduino breath (I presume).

```
        ... UsbConnection and UsbDevice init stuff ....
        serial = UsbSerialDevice.createUsbSerialDevice(device, usbConnection)

        if (serial.syncOpen()) { // setup for ch340
            serial.setBaudRate(9600) 
            serial.setDataBits(UsbSerialInterface.DATA_BITS_8)
            serial.setParity(UsbSerialInterface.PARITY_ODD)
            serial.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF)
             
            //
            // here be dragons! don't let the "syncOpen" name fool you.
            //
            Thread.sleep(2000); // YMMV with different chips
            //  
            // without this delay the write below does not occur. 
            //    

            serial.syncWrite(byteArrayOf(yourMagicByteValues), timeout);
         }
```

Might be a good point to mention in the docs. Thanks for the great lib!

felHR85, 2016-09-02:
Hello @utkukaratas 
That's probably because Arduino firmware just wait some time to receive a signal that means a new sketch is going to be uploaded. If that signal doesnt happen it will load the last sketch.

I will add that to the example. Thanks!

Vijayapriyacs, 2017-06-28:
serialPort = UsbSerialDevice.createUsbSerialDevice(device, connection);
                    if (serialPort != null) {
                        if (serialPort.open()) { //Set Serial Connection Parameters.
                            serialPort.setBaudRate(9600);
                            serialPort.setDataBits(UsbSerialInterface.DATA_BITS_8);
                            serialPort.setStopBits(UsbSerialInterface.STOP_BITS_1);
                            serialPort.setParity(UsbSerialInterface.PARITY_NONE);
                            serialPort.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                            try {
                                Thread.sleep(1000);
                            } catch (InterruptedException e) {
                                e.printStackTrace();
                            }
                            serialPort.write("Connected".getBytes());//not able to write here
                            try {
                                Thread.sleep(3000);
                            } catch (InterruptedException e) {
                                e.printStackTrace();
                            }
                            serialPort.write("start".getBytes());// not able to write here
                            initdone=true;
                      }