felHR85/UsbSerial #76: Does not recieve sometimes

anugrah8, 2016-09-23:
Works fine, but sometimes 
1- when I connect USB to android device it shows 'USB Ready' in toast (after I allow it to use USB Host) but dose not receive anything.
2-And then I remove the OTG connector form device, the toast should display 'USB disconnected' but it doesn't displayed. 
3-Now I connect USB again, now toast display 'USB disconnected' and then ask for permission to use USB Host.
- I'm using USB to OTG convertor cable, at point 2 if I disconnect the USB from convertor then also nothing has happen.  

* One thing I also notice that when the connection established, android device receiving data and now if I turn OFF and then ON the transmitting device, the android device stops receiving and the same condition occur as point 2 above.   

Thanks.

gauravbrd, 2016-10-18:
It depends on how you have programmed for starting the USB serial communication. USB connected and Disconnected toasts are just Device attach messages from Android Itself but It has nothing to do with Serial start. Does your GUI has anything which Listens USB on receive callback ?

I am using  simple GUI, when I press start button it will go and establish serial communication.
` HashMap<String, UsbDevice> usbDevices = usbManager.
                                     getDeviceList();
                             if (!usbDevices.isEmpty()) {
                                 boolean keep = true;
                                 for (Map.Entry<String, UsbDevice> entry : usbDevices.entrySet()) {
                                     device = entry.getValue();
                                     int deviceVID = device.getVendorId();
                                     if (deviceVID == 0x2341 || deviceVID == 0x2A03)//Arduino Vendor ID //2A03, 2341
                                     {
                                         PendingIntent pi = PendingIntent.getBroadcast(this, 0, new Intent(ACTION_USB_PERMISSION), 0);
                                         usbManager.requestPermission(device, pi);
                                         keep = false;
                                     } else {
                                         connection = null;
                                         device = null;
                                     }
                                     if (!keep)
                                         break;
                                 }
                             }`

Vijayapriyacs, 2017-06-08:
I too have  problem with  serial port write
1.First i connected usb serial port then i called serialport.write("firststring".getBytes()) one by one serialport.write("second string".getBytes());  like  this.
2.Problem i faced is serial.read(callback) only shows "firststring". "Secondstring "that i wrote not shown by serial read() 
how can i solve this problem