felHR85/UsbSerial #77: Could not receive full data bytes.

williamtran1988, 2016-10-10:
Hi everyone,

I am using this library to read data from EnOcean USB 300 device:
https://www.enocean.com/en/enocean_modules/usb-300-oem/. I am trying to receive press and release data from Button which is connect with this USB via Radio Signal.

I am facing an issue that could not receive enough data bytes:

                    serialPort.setBaudRate(57600);
                    serialPort.setDataBits(UsbSerialInterface.DATA_BITS_8);
                    serialPort.setStopBits(UsbSerialInterface.STOP_BITS_1);
                    serialPort.setParity(UsbSerialInterface.PARITY_NONE);

at data callback: onReceivedData

As usb button specification, i will receive 21byte for each time i pressed and released button. However, sometimes i could not receive full data byte as documentation. 

I have a solution for this: 
private UsbSerialInterface.UsbReadCallback mCallback = new UsbSerialInterface.UsbReadCallback() {
        @Override
        public void onReceivedData(byte[] arg0) {

            if(arg0.length < 21){
                if(dataReceived == null){
                    dataReceived = new byte[0];
                }
                byte[] destination = new byte[dataReceived.length + arg0.length];
                System.arraycopy(dataReceived, 0, destination, 0, dataReceived.length);
                System.arraycopy(arg0, 0, destination, dataReceived.length, arg0.length);
                dataReceived = destination;
            }
            else{
                dataReceived = arg0;
            }
            if(dataReceived.length == 21){
                byte[] senderID = Arrays.copyOfRange(dataReceived, 9, 12);
                int value = dataReceived[7];

                Intent intent = new Intent(ACTION_USB_DEVICE_PRESSED);
                intent.putExtra("id", hexToString(senderID));
                intent.putExtra("value", value);
                sendBroadcastMessage(intent);

                dataReceived = null;
            }
        }
    };

But it looks like wrong logical, because i could see data is not mapped correct when multi button pressed or release in the same time. Data is mixed together, therefore i could not parse data as expected.

Could somebody help me on this?

williamtran1988, 2016-10-11:
finally, i found out the answer by myself:

https://github.com/felHR85/UsbSerial/issues/37

felHR85, 2016-10-11:
Nice!

closed 2016-10-11 by felHR85