felHR85/UsbSerial #83: UsbHost FTDI device not detected

HappyDr0id, 2016-11-25:
Hello everyone,

I am currently facing a problem about making an Android device recognize a FTDI Device.

I have an Arduino Pro Mini connected to my Android smartphone (which is Android 6.x) through a FTDI cable (using a FT232RQ chip : https://www.sparkfun.com/products/9717) and a USB<-->microUSB OTG cable.
The Android device is not raising the ACTION_USB_ATTACHED action through the Broadcast receiver when I plug the Pro Mini, and calling UsbManager.getDeviceList() is not giving anything either. In comparison, the USB detection and communication is perfectly working with an Arduino UNO board (USB-B to USB-A cable and the same OTG cable).

It seems that @midhunm45 is facing the same issue : https://github.com/felHR85/UsbSerial/issues/81 but there are no answers for the moment.

Does anyone have any clue about making the connection working with a FTDI cable ? Is it mandatory to use such controller chip as a FT311D (http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT311D.pdf) to make the transition between FTDI and the USB OTG cable ?

Thanks in advance guys !

mifkamaz, 2017-10-03:
Found a solution?

closed 2018-12-31 by felHR85

Chidhambararajan, 2019-12-17:
why was this issue closed? Was a fix obtained

Looks like the FTDI chip is not getting recognized in when the port is not in otg mode.
I have tried out a couple of apps which implements the same library.All of those apps 
are not recognizing when tried through a ftdi cable which doesn't activate OTG mode

Is it that FTDI devices can't communicate with smartphone if OTG Mode isnt activated?