felHR85/UsbSerial #87: Problem in receiving data

saury001, 2016-12-31:
Whenever I am receiving data in UsbReadCallback and displaying it in setText() method of textview its showing incorrect values i.e. a part of value eg. Sometimes it display 00 for 100 and 2 for 42 etc. But its working fine in append() method of textview.What to do?

dani2584, 2018-07-23:
I have Same Problem. that's issue so important. please patch it

jamesjude123, 2018-10-01:
same problem here.

jamesjude123, 2018-10-01:
hey @saury001 , i have a problem in the receiving data, how did you manage it? because mine I cant receive any , can you help me?

here is my code:

```
 public void openDevice()
    {
        connection = usbManager.openDevice(printer);
        debug.append("printer:  "+printer +"\n");
        debug.append("connection:  "+connection+"\n");
        serialDevice = UsbSerialDevice.createUsbSerialDevice(printer, connection);
        debug.append("serialDevice:  "+serialDevice.toString()+"\n");
        if (serialDevice != null) {
            if (serialDevice.open()) {
                serialDevice.setBaudRate(9600);
                serialDevice.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialDevice.setParity(UsbSerialInterface.PARITY_NONE);
                serialDevice.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialDevice.setRTS(true);
             //   serialDevice.read(mCallback);
            } else {
                debug.append("Printer port not open \n");
            }
        } else {
            debug.append("Serial is null \n");
        }
    }

    public void sendData(String data)
    {
        debug.append("\n sendData  \n"+data.getBytes());
        serialDevice.write(data.getBytes());
        serialDevice.read(mCallback);

    }

    String respond;
    private UsbSerialInterface.UsbReadCallback mCallback = new UsbSerialInterface.UsbReadCallback() {
        @Override
        public void onReceivedData(byte[] arg0)
        {
            Toast.makeText(getApplication(),"RESPONSE TERMINAL: "+arg0,Toast.LENGTH_LONG ).show();
            debug.append("RESPOND TERMINAL ONLY"+respond+"\n");

        }
    };
 
```

is there any i missed ? can you help me on this please?