felHR85/UsbSerial #88: How to communicate with more than one USB Serial devices

suyog44, 2017-01-15:
I am trying to make android code work which can communicate with multiple USB serial devices and android tablet.

I am able to communicate with single USB device at a time. but, now when I try to connect and communicate with multiple USB serial device App is failing.

jiuwu0905, 2017-02-06:
I am also look for the solution of can not communicate multiple USB serial devices.

jamesjude123, 2018-10-01:
public void openDevice()
    {
        connection = usbManager.openDevice(printer);
        debug.append("printer:  "+printer +"\n");
        debug.append("connection:  "+connection+"\n");
        serialDevice = UsbSerialDevice.createUsbSerialDevice(printer, connection);
        debug.append("serialDevice:  "+serialDevice.toString()+"\n");
        if (serialDevice != null) {
            if (serialDevice.open()) {
                serialDevice.setBaudRate(9600);
                serialDevice.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialDevice.setParity(UsbSerialInterface.PARITY_NONE);
                serialDevice.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialDevice.setRTS(true);
             //   serialDevice.read(mCallback);
            } else {
                debug.append("Printer port not open \n");
            }
        } else {
            debug.append("Serial is null \n");
        }
    }

felHR85, 2019-05-01:
[Added Multiple Serial port Api](https://github.com/felHR85/UsbSerial/wiki/7.-Multiple-Serial-ports)

closed 2019-05-01 by felHR85