felHR85/UsbSerial #91: Seem to be having some issues with Baud Rate

skipe777, 2017-02-10:
I have been attempting to use the UsbSerial library on Android (Nexus 7 with Marshmallow). I have an FTDI Usb to 232 cable connected up through a OTG cable. The other end is an ODROID C2 running Ubuntu Linux. I have tested the cable and the connection to the linux machine up to a Baud Rate of 230400. I have been trying to get the Android->Linux connection working at 115200, but haven't been able to get data over 9600. Once I turn it up to anything higher I no longer see it on the other end (or its just garbage).

I am operating in non-canonical / raw input on the other side so I am just reading single bytes. I guess I should also note that I am seeing echoing on the line on the Android side and I have tried both Async and the Sync calls.

Has anyone used the library at 115200 (or higher) with any success? I am using the example code provided in for the Sync calls at this point. Is anyone else seeing any echoing of their own sent data?

I can provide the Linux code if necessary, but its pretty straight forward C++ that I have been using for years on other projects.

AdrianDC, 2017-02-19:
Same issue on my setup while I was trying to move away from Physicaloid.

I'm working on Android serial communication with an ARM mbed 1768,
and I can configure the baudrate dynamically on both sides.

The library doesn't apply the baudrate requests,
it is always stuck on 9600 although the CDC should be 115200 by default.

jzaruba, 2017-04-21:
Same here. Using putty I can talk to my Arduino at whatever baud-rate. Using the library I am stuck at 9600.

jzaruba, 2017-05-06:
Turns out the issue I was facing seems to be non-present when using genuine Arduino Uno, rather than a clone I was using before that. The question is whether the issue lies within the clone hardware or the class handling its chip: CH34xSerialDevice.java.

jzaruba, 2017-05-06:
Tested with geniune Arduino/CDCSerialDevice and a clone NHduino/CH34xSerialDevice. Apparently USBSerial can only talk to hose only at whatever is the default baud rate in their respective Java classes: 9600 for the clone and 115200 for Arduino. (Admittedly, with the Arduino I only tested 115200 and 230400.)

jzaruba, 2017-05-07:
I can attest that modifying default baud rate in CDCSerialDevice.java worked for me. Now my Android app talks to (genuine) Arduino working at baud rate of 230400. You need to modify different class when using a clone, most likely CH34xSerialDevice.java.

closed 2019-04-21 by felHR85