felHR85/UsbSerial #92: CDC Device receiving data problem

snowbizz84, 2017-02-12:
Hi,
I observed a problem with receiving asynchronously data from CDC device (Arduino Mega ADK with Atmega 16u2 communication chip).
The data which comes from Serial is divided into several parts:
1) Number of bytes obtained from Read buffer: 1
2) Number of bytes obtained from Read buffer: 48
3) Number of bytes obtained from Read buffer: 21

but on the other devices (PL2303, CH340) everything is OK and data comes as a one package:
1) Number of bytes obtained from Read buffer: 70

Port configuration:
port.setBaudRate(115200);
port.setDataBits(UsbSerialInterface.DATA_BITS_8);
port.setStopBits(UsbSerialInterface.STOP_BITS_1);
port.setParity(UsbSerialInterface.PARITY_NONE);
port.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);

I think that the lower speed I set, the more slices I get.

Regards,
MK

Gupta126, 2017-05-02:
@snowbizz84  did you get any solution on this ? can you please tell me how you persist connection between device and android app ?

snowbizz84, 2017-05-02:
@Gupta126 I have only a dummy solution -> I send a length of data frame as the first two bytes in each frame and I am waiting and reading data until I receive all bytes.

Gupta126, 2017-05-02:
can you please tell me how you connected device right now i am not able to even connect device its not founding device @snowbizz84

mahermeg17, 2017-11-29:
Hi,
Did anyone found a solution for this problem? I have the same when reading from UART port of the Pico PI IMX6UL.
Using the sync or async mode doesn't make a change.
The some problem is invoked here https://github.com/felHR85/UsbSerial/issues/38
and the main developer said it would be a hardware issue.

mahermeg17, 2017-11-29:
I didn't have this problem when using https://github.com/mik3y/usb-serial-for-android

closed 2018-12-31 by felHR85

snowbizz84, 2019-01-01:
Is this bug fixed?