felHR85/UsbSerial #95: USB Permission - save "allow" status

Vikramsinh, 2017-03-15:
Hello, My problem is, I want to save "Permission Allow" status of USB Permission when permission comes in front of user. and next time I don't want to show again that dialog and send connected flag to device.

JugeHuge, 2017-10-05:
If have can make custom rom image. Then you could make modification to UsbPermissionActivity.java
i.e. in Marshmallow AOSP release on USBPermissionsActivity.java comment out setupAlert(); and add finish(); then check that finish() method and add mPermissionGranted = true; at the beginning.

abhishekkumar1902, 2017-11-19:
Did you got any resolution?
This app every time asks for permissions, is there a way to make it only once when the USB is attached?

lotfiGipsyKing, 2018-01-23:
the same problem did you found something,
i get the popup permission every time the USB is attached

JugeHuge, 2018-01-24:
Intent filter is one way to go. It will ask only once if save setting in query.
https://stackoverflow.com/questions/15255524/android-usb-device-permission-and-intent-filter

lotfiGipsyKing, 2018-01-30:
Thank you for the answer,, after make the change (stackover link), i still have the popup, should i delete the fonction requestpermission() in the USBService.java ???

eyedol, 2018-01-30:
Make sure to see if the USB device has been granted permission before requesting for another. The pseudo snippet code below should help. Also, make sure you've setup USB device filtering as suggested in the SO link above or as documented [here](https://developer.android.com/guide/topics/connectivity/usb/host.html)

```java
if (usbManager.hasPermission(device)) {
  // perhaps open a usb communication?
} else {
  requestUserPermission()
}
```

lotfiGipsyKing, 2018-01-30:
I tried this but he doesn't work for me, i think is a issue in the android UsbPermissionActivity class as described here : https://stackoverflow.com/questions/13726583/bypass-android-usb-host-permission-confirmation-dialog ,

lotfiGipsyKing, 2018-01-30:
thank you, it worked for me.

kamrankhorasani, 2019-07-08:
> Make sure to see if the USB device has been granted permission before requesting for another. The pseudo snippet code below should help. Also, make sure you've setup USB device filtering as [...]

i want to use this option for check permission without any intent filter is it possible to save permission ?