felHR85/UsbSerial #96: Receiving data multiple time and string value is not accurate

animeboy666, 2017-03-17:
Hello 
 I have an issue in reading data which i write from arduino side. For example i'm trying to send a string value like "P152" but in android side it receives in pieces like "p","152","p152" or "P1","52","p152" like this, also there are some delay. if my string length is "AP152Q524" then it took around 28-30 second to fetch complete string.

I don't understand what i do to deal with this so please help me..  

Thank you.

snowbizz84, 2017-03-19:
Which chip you are using?

dani2584, 2018-07-23:
I have same Problem. what's Problem?

kjitender469, 2019-02-16:
I am also facing the same issue? Can anyone help me on this ?

spss20, 2021-06-18:
I am also facing same issue @felHR85 I am receiving my packets in parts in delay of 20 milliseconds , I am not able to parse my full packet without using custom function to merge packets with delay less then 20 milliseconds.