felHR85/UsbSerial #97: App crash when "createUsbSerialDevice"

williamtran1988, 2017-03-24:
Hi All, 

I am having an issue when integrate with library. My application was crashed when "createUsbSerialDevice", 

i have 2 case as below:
- Plug USB to android device: ==> OK
   + App is open because i have device filter USB attached.
   + Can create a connection and open it.
- USB is already plugged.
   + Reboot android device.
   + Permission is granted.
   + openDevice success
   + createUsbSerialDevice ===> FAILED.

Could somebody help me on this?

PawelSzymanski89, 2018-10-29:
i have simmilar issue...

felHR85, 2018-11-08:
@PawelSzymanski89 Some info from Logcat would be great

felHR85, 2018-11-18:
Hi @ToninoTarsi 
What Android version do you have? Also It would be great to take a look of the exception stacktrace

ToninoTarsi, 2018-11-18:
Thanks Felipe
Android 6 ( LG G3 ) . For the stacktrace I will be happy to be able to provide you one but I have no idea on how to connect simultaneously the FDTI and the PC to my phone.  How do you do that ?
Regards

felHR85, 2018-11-18:
Android 6 should work fine. What kind of UsbSerial device are you connecting? if createUsbSerialDevice is returning null that means the usb device "is not supported" because VID-PID are not recognized.
[For debugging through wifi try this](http://stuffandtech.blogspot.com/2012/03/android-quick-tip-adb-over-wifi.html)

ToninoTarsi, 2018-11-18:
Thanks for the information about ADB ove WiFi  .. I will try. The device is an  FTDI ( device.getVendorId() return 1027 ) and your example sync works fine so it is supported and your library works fine.
The ADB over wifi will help to understand what I'm missing .. I hope.
Thanks

felHR85, 2018-11-18:
@ToninoTarsi And the product id? Vendor id is standard FTDI but maybe that product id is not supported.

ToninoTarsi, 2018-11-18:
PID = 24597 (dec )  0x6015

felHR85, 2018-11-18:
@ToninoTarsi It should be supported

ToninoTarsi, 2018-11-18:
Yes .. it works. It was my fault. Thanks for indicating me how to use ADB over Wifi .. I did not known that :-)
Regards

closed 2018-12-26 by felHR85