felHR85/UsbSerial #99: 10xCH340 via Hub - tablet reboots on async read

todor-simeonov, 2017-04-11:
I'm working on a project of a testing machine with 10 devices and a tablet for a HMI (human-machine interface). I use a 10 port USB hub, actually inside there are three stacked 4-port hubs using no-name chip on board. Usb-serial devices are CH340. The tablet is a no-name MT8127 with Android 4.4.2, Kernel 3.4.67.

Connecting each of the CH340 devices through the hub works fine. I'm able to send an receive with no issue. I request and receive permission for all devices step by step - a new request is made when the previous one has received permission. So far it works well. The problem comes out when I try to open all 10 devices simultaneously with asynchronous read. The first 4 ports are opening OK. When I try to open the 5-th port the whole Android device reboots.

I'm debugging the issue but it goes very slow because of each time rebooting, reconnecting ADB in TCP/IP mode an so on.

As far as I know the reboot comes in file **UsbSerialDevice.java**, method **read()**, line:

`workerThread.getUsbRequest().queue(serialBuffer.getReadBuffer(), SerialBuffer.DEFAULT_READ_BUFFER_SIZE);
`

Do you have any clue what may cause the reboot?

todor-simeonov, 2018-10-30:
Thanks for the answer. No there are no current issues, because the USB hub has external power supply. The problem still exists ( 1.5Y later :) ) but I did a workaround for my application to use lower count USB modules.

felHR85, 2018-12-26:
Added [multiple serial ports api](https://github.com/felHR85/UsbSerial/wiki/7.-Multiple-Serial-ports)

closed 2018-12-26 by felHR85