felHR85/UsbSerial #103: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.felhr.usbserial.UsbSerialDevice.write(byte[])' on a null object reference

Vijayapriyacs, 2017-05-18:
When ever  I try to use  serial port.write("DATA".getbytes) 
i got Following issue :java.lang.NullPointerException: Attempt to invoke virtual method 'void com.felhr.usbserial.UsbSerialDevice.write(byte[])' on a null object `reference```
//override method called from fragmentA
 public void send(String s){
       Log.d("From fragement A",s);
        serialPort.write("DATA".getBytes());
               }
//
 private final BroadcastReceiver broadcastReceiver = new BroadcastReceiver() { //Broadcast Receiver to automatically start and stop the Serial connection.
        @Override
        public void onReceive(Context context, Intent intent) {
            if (intent.getAction().equals(ACTION_USB_PERMISSION)) {
                boolean granted = intent.getExtras().getBoolean(UsbManager.EXTRA_PERMISSION_GRANTED);
                Toast.makeText(context, "boolean" + granted, Toast.LENGTH_LONG).show();
                if (granted) {
                    connection = usbManager.openDevice(device);
                    Toast.makeText(context, "connection" + connection, Toast.LENGTH_LONG).show();
                    serialPort = UsbSerialDevice.createUsbSerialDevice(device, connection);
                    if (serialPort != null) {
                        if (serialPort.open()) { //Set Serial Connection Parameters.
                            Toast.makeText(context, "serialopened", Toast.LENGTH_LONG).show();
                            serialPort.setBaudRate(9600);
                            serialPort.setDataBits(UsbSerialInterface.DATA_BITS_8);
                            serialPort.setStopBits(UsbSerialInterface.STOP_BITS_1);
                            serialPort.setParity(UsbSerialInterface.PARITY_NONE);
                            serialPort.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                        } else {
                            Log.d("SERIAL", "PORT NOT OPEN");
                            Toast.makeText(context, "PORT NOT OPEN", Toast.LENGTH_LONG).show();
                        }
                    } else {
                        Log.d("SERIAL", "PORT IS NULL");
                        Toast.makeText(context, "PORT IS NULL", Toast.LENGTH_LONG).show();
                    }
                } else {
                    Log.d("SERIAL", "PERM NOT GRANTED");
                    Toast.makeText(context, "PERM NOT GRANTED", Toast.LENGTH_LONG).show();
                }
            } else if (intent.getAction().equals(UsbManager.ACTION_USB_DEVICE_ATTACHED)) {
              serialPort.open();
            } else if (intent.getAction().equals(UsbManager.ACTION_USB_DEVICE_DETACHED)) {
               serialPort.close();

            } else if(intent.getAction().equals(CUSTOM_INTENT)){
                serialPort.close(); Toast.makeText(getBaseContext(), "HEY BUDDY supper it works", Toast.LENGTH_SHORT).show();}
           
        }
    };

TanayaBhave, 2017-09-20:
Hey did you find a solution to this ? I am facing the same issue.

Vijayapriyacs, 2017-09-21:
You have to check serial object not to be null before using write (byte[]).

Vijayapriyacs, 2017-09-21:
Also use some delay before every continuous write.

TanayaBhave, 2017-09-21:
can you explain what variable is your serial object in the above code. So i get an idea as to what should I check. Sorry for the stupid questions.. I am a beginner !

Vijayapriyacs, 2017-09-21:
SerialPort is variable that  i used as serial object

closed 2018-10-28 by felHR85