felHR85/UsbSerial #111: usb device not supported

sowbarsathik, 2017-07-11:
usb serial library is not supporting the tablet 'RTC-900R' Please check the issue and provide the solution.

mws-rmain, 2019-01-11:
Check #193.
One of my test devices (a Panasonic ToughPad FZ-B2) had a embedded USB-connected WiFi adapter. When android supplied a list of connected USB devices, they appear in random order. With the possible exception of the multi-device code, the examples in this repo only attempt to link to the first (non-hub) device in the list. If this was the USB-WiFi in my tablet, I'd get 'not supported'. If it happened to be the serial adapter, everything worked fine.

Hopefully this is also your issue, and #193 will fix it for you too.

felHR85, 2019-01-11:
#193 Merged to master. Please give it a try

closed 2019-01-11 by felHR85