felHR85/UsbSerial #112: Working w/ virtual ports

russellhoff, 2017-07-17:
Hi!

I've some doubt about how to use this library. Currently I'm testing a simple project, which I'm gonna explain now.

I've got an android app working with your library. This app'd wait for some data, sent by serial port. That's okay.

I want my computer to emulate serial ports and send data using hyperterminal.

Well, I've downloaded some software from Eltima: [Virtual Serial Port Driver ](https://www.eltima.com/products/vspdxp/) to create a couple of virtual serial ports, and [Advanced Serial Port Terminal](https://www.eltima.com/products/serial-port-terminal/).

Everything works great creating a session port COM1 in terminal (to send text) and creating session port COM2 in terminal (to receive text).

Then, I left only one connection in eltima advanced serial port terminal and opened the android app... But within the app, I get the toast messasge "USB not supported".

Am I missing some configuration? What am I doing wrong?

Thank you for your support.

felHR85, 2018-10-28:
Hi! @russellhoff 
Sorry for the delayed response. I have no idea what that didnt work. Did you solve it?

russellhoff, 2018-10-29:
Hi! Don't worry. I left the implementation on Android and started an application on Spring Boot.
Thanks!

closed 2018-10-29 by russellhoff