felHR85/UsbSerial #116: FTDI

z0rbal, 2017-08-08:
Hi, I've been using two FT232s thru android OTG (with hub) with your code to connect to some dedicated  hardware (NMEA on port1 and my own hardware on port2)  - it's working great. However I want to use the dual FT2232H device (so there's no hub) - it reports as having two interfaces - but I guess your lib doesn't handle it - any idea how to do it? Also is there a callback for writebuffer remaining or a count of characters sent.
Thanks.

bill-chadwick, 2017-11-10:
You just need to enumerate the interfaces on the usb device and pass the interface index into the specialised createUsbSerialDevicefunction that accepts an extra parameter - UsbSerialDevice.createUsbSerialDevice(device, connection, usbInterface);

z0rbal, 2017-11-14:
Thanks Bill - I knew I was missing something - I'll give it a go!

closed 2018-10-28 by felHR85