felHR85/UsbSerial #118: Input stream

cleevie, 2017-08-28:
Is there a way to create a inputstream from the data being recieved from a serial device?
The device i am using just spews out data constantly no need communicate back or anything

felHR85, 2017-08-28:
Hello @cleevie 
Try this classes: https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/SerialInputStream.java

https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/SerialOutputStream.java

bdureau, 2017-10-04:
Hello
I am also using inputStream because I want to use it the same way I am using the Android bluetooth connection. I have have reused your SerialInputStream class, modified it and overwrite the available method to return the size. It partially works but crashes if to much data.
Here is the class
Can you help please.
Regards
Boris
    public class SerialInputStream extends InputStream implements UsbSerialInterface.UsbReadCallback
    {
        protected final UsbSerialInterface device;
        protected ArrayBlockingQueue data = new ArrayBlockingQueue<Integer>(256);
       
        protected volatile boolean is_open;

        public SerialInputStream(UsbSerialInterface device)
        {
            this.device = device;
            is_open = true;
            device.read(this);
        }

        @Override
        public int available(){
            return data.size();
        }

        @Override
        public int read()
        {
            while (is_open)
            {
                try
                {
                    return (Integer)data.take();
                } catch (InterruptedException e)
                {
                    // ignore, will be retried by while loop
                }
            }
            return -1;
        }

        public void close()
        {
            is_open = false;
            try
            {
                data.put(-1);
            } catch (InterruptedException e)
            {
                e.printStackTrace();
            }
        }

        public void onReceivedData(byte[] new_data)
        {
            for (byte b : new_data)
            {
                try
                {
                    data.put(((int)b) & 0xff);
                } catch (InterruptedException e)
                {
                    // ignore, possibly losing bytes when buffer is full
                }
            }
        }
    }

bdureau, 2017-10-05:
I have tracked and found why it was crashing. This is due to missing piece of data. I am using 3DR radio module over the Android USB so I am not sure if it is the radio module losing data or a problem with the Inputstream.
I will do some additionnal investigations and post the results here.

bdureau, 2017-10-06:
My issue is now fixed, this was due to the 3DR module and not the usb.
However you need to replace the SerialInputStream  with my code so that the available function works otherwise it will return 0.
I will publish an example using an input stream

cleevie, 2017-10-25:
I am trying to do the following

SerialInputStream serialInput = new SerialInputStream(serialPort);

AisReader aisReader = AisReaders.createReaderFromInputStream(serialInput);

                    aisReader.registerHandler(new Consumer<AisMessage>() {
                        @Override
                        public void accept(AisMessage aisMessage) {
                            if (mHandler != null)
                            mHandler.obtainMessage(MESSAGE_FROM_SERIAL_AIS, aisMessage).sendToTarget();
                        }
                    });
                    aisReader.start();
                    try {
                        aisReader.join();
                    } catch (InterruptedException e) {
                        e.printStackTrace();
                    }

the ais reader just needs the inputstream to out data. Am i possibly doing this incorrectly?

bdureau, 2017-10-26:
Are you overwridding the SerialInputStream  like I did? 
There seam to be an issue with the SerialInputStream  in the Android usb as it does not count the number of elements

bdureau, 2017-10-26:
I have a usage exemple in my repository
https://github.com/bdureau/BearConsole2

jamesjude123, 2018-10-01:
@bdureau  hey brother, how did you manage the data received data ?

i already setup the open and connect and sending serial data, but the problem i cant received a response,
how did you manage this?

```
  public void openDevice()
    {
        connection = usbManager.openDevice(printer);
        debug.append("printer:  "+printer +"\n");
        debug.append("connection:  "+connection+"\n");
        serialDevice = UsbSerialDevice.createUsbSerialDevice(printer, connection);
        debug.append("serialDevice:  "+serialDevice.toString()+"\n");
        if (serialDevice != null) {
            if (serialDevice.open()) {
                serialDevice.setBaudRate(9600);
                serialDevice.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialDevice.setParity(UsbSerialInterface.PARITY_NONE);
                serialDevice.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialDevice.setRTS(true);
             //   serialDevice.read(mCallback);
            } else {
                debug.append("Printer port not open \n");
            }
        } else {
            debug.append("Serial is null \n");
        }
    }

    public void sendData(String data)
    {
        debug.append("\n sendData  \n"+data.getBytes());
        serialDevice.write(data.getBytes());
        serialDevice.read(mCallback);

    }

    String respond;
    private UsbSerialInterface.UsbReadCallback mCallback = new UsbSerialInterface.UsbReadCallback() {
        @Override
        public void onReceivedData(byte[] arg0)
        {
            Toast.makeText(getApplication(),"RESPONSE TERMINAL: "+arg0,Toast.LENGTH_LONG ).show();
            debug.append("RESPOND TERMINAL ONLY"+respond+"\n");

        }
    };
```

is there any missing or mistake that i did? please help me

bdureau, 2018-10-01:
@jamesjude123 you need to overwrite the SerialInputStream class (replace it with mine)
Check my code on my github to have an example