felHR85/UsbSerial #121: Waiting for a blocking GC (Examplesync)

CvdH, 2017-09-26:
**Tested on a Huawei Nexus 6P running Android 8.0.0 connected to Wemos D1 mini pro (CP2104 Usb to Uart)**

I copied the [SerialPortExampleSync](https://github.com/felHR85/UsbSerial/tree/master/examplesync/src/main/java/com/felhr/serialportexamplesync) almost in its entirety and only added a few buttons that sent a different String.

The reading and writing over the USB otg cable in the app seems to go without any issues, correct data is sent and received. My problem is that after a seemingly random time (ranging from immediately to after about a minute) the app will become completely unresponsive (except for the keyboard UI if text was being entered at the time) and the following messages will appear in the Logcat window:

`I/zygote64: Waiting for a blocking GC ObjectsAllocated`
`I/zygote64: Thread[3,tid=9450,WaitingInMainSignalCatcherLoop,Thread*=0x75fa4bf400,peer=0x13040088,"Signal Catcher"]: reacting to signal 3`
`I/zygote64: Waiting for a blocking GC ObjectsAllocated`
`I/libc: Requested dump for tid 9444 (sper.androidapp)`

While monitoring Memory usage does not seem to spike, indicating that a memory leak might not be the problem?

I added an obscene amount of debug logging statements to the code but despite that I cannot find any code that might be causing an infinite loop or memory leak.

I am out of ideas of how to fix this. If the fault is not with the library but in my own code, what could be causing this?

AlonShahaf, 2018-03-21:
@CvdH hey, although we're not using this library, we encounter a similar problem when connecting a USB device to our app.
Did you find any solutions or reasons for this to happen?

Thanks in advance.

closed 2018-10-28 by felHR85

LJ-SH, 2019-09-04:
@felHR85 I hit the same issue when I used your library. Can you please advice why you close the issue? It's been fixed in new release. Thanks in advance.

davidgiga1993, 2019-10-01:
I'm facing the same issues with a completely unrelated project (also using USB data transfer) but can someone please give some insights of what fixed the issue?
@felHR85 maybe?

steroom, 2020-07-24:
@davidgiga1993 Did you or anyone else ever get this resolved? Same situation as you, an unrelated project with USB data transfer, and again looking for any insights. Any help would be much appreciated.

spss20, 2020-07-25:
@steroom Hey buddy, my issue is solved. No more Gcblocker. Change the timeout in RunnerThread of service. I have closed the project a month ago , so i dont remember the exact function name. But ya global search for timeout and change the timeout to 2000 or 1000 milliseconds. I hope it will help you.

steroom, 2020-07-27:
@spss20 Thanks for the tip. Tried updating timeout, same GC headache. Issue only occurs on android 9 PIE, implementation has been stable since android 5...something that I can't find has changed in the USB implementation for PIE. Of course this is quite specific issue but this is the only thread anywhere that came close to the issue of GC blocking and USB connection seeming to be culprit.