felHR85/UsbSerial #123: hex data greater than 7f

NetAfarin, 2017-09-30:
i cant get data from serial with greater than 7f in "handle message" object that have string's behavior.
i changed charset to "UTF-8" , "UTF-16" ,  "ISO-8859-1" and another things but it still did not make any difference.

```
private static class MyHandler extends Handler {
        private final WeakReference<Main2Activity> mActivity;

        public MyHandler(Main2Activity activity) {
            mActivity = new WeakReference<>(activity);
        }

        @Override
        public void handleMessage(Message msg) {
            switch (msg.what) {
                case UsbService.MESSAGE_FROM_SERIAL_PORT:
                    String data = (String) msg.obj;
                    mActivity.get().textLog.append(bytesToHex(data.getBytes()));

```
the bytesToHex function:

```
public static String bytesToHex(byte[] bytes) {
        char[] hexChars = new char[bytes.length * 2];
        for (int j = 0; j < bytes.length; j++) {
            int v = bytes[j] & 0xFF;
            hexChars[j * 2] = hexArray[v >>> 4];
            hexChars[j * 2 + 1] = hexArray[v & 0x0F];
        }
        return new String(hexChars);
    }
```

Aerotic, 2017-11-01:
I‘ve met  the same question.My solution is to divide a 8 bit hex data into to two parts in two char variables.
It did work. But I also need a better way.

coinageboy, 2017-12-11:
Aerotic,
excellent comment - did resolve my problem.

jamesjude123, 2018-10-01:
hey brother, can you tell me what you did how did you get the response from the terminal ?

because , i did not receive response data,

```
 public void openDevice()
    {
        connection = usbManager.openDevice(printer);
        debug.append("printer:  "+printer +"\n");
        debug.append("connection:  "+connection+"\n");
        serialDevice = UsbSerialDevice.createUsbSerialDevice(printer, connection);
        debug.append("serialDevice:  "+serialDevice.toString()+"\n");
        if (serialDevice != null) {
            if (serialDevice.open()) {
                serialDevice.setBaudRate(9600);
                serialDevice.setDataBits(UsbSerialInterface.DATA_BITS_8);
                serialDevice.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialDevice.setParity(UsbSerialInterface.PARITY_NONE);
                serialDevice.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialDevice.setRTS(true);
             //   serialDevice.read(mCallback);
            } else {
                debug.append("Printer port not open \n");
            }
        } else {
            debug.append("Serial is null \n");
        }
    }

    public void sendData(String data)
    {
        debug.append("sendData  \n"+data.getBytes());
        serialDevice.write(data.getBytes());
        serialDevice.read(mCallback);

    }

    String respond;
    private UsbSerialInterface.UsbReadCallback mCallback = new UsbSerialInterface.UsbReadCallback() {
        @Override
        public void onReceivedData(byte[] arg0)
        {
            Toast.makeText(getApplication(),"RESPONSE TERMINAL: "+arg0,Toast.LENGTH_LONG ).show();
            debug.append("RESPOND TERMINAL ONLY"+respond+"\n");
            try {

                respond = new String(arg0, "US-ASCII");

            } catch (UnsupportedEncodingException e) {
                e.printStackTrace();
                debug.append("ERROR"+e.toString()+"\n");
            }

        }
    };
```

this is my code can you hgelp me guys ?
@coinageboy  @Aerotic  and @NetAfarin

closed 2018-10-28 by felHR85