felHR85/UsbSerial #126: Trouble with dual RS232 FTDI device

bill-chadwick, 2017-11-10:
I find the library works fine with single RS232 adapters using FTDI chips. 

We have just got a 2 port StarTech adapter - one Usb Device with two RS232 interfaces. The felhr library works a bit, and can connect to both ports, but after a while, with just the first port in use, I get

 FATAL EXCEPTION: Thread-16
                                                            Process: com.me.test, PID: 9228
                                                            java.lang.ArrayIndexOutOfBoundsException: length=61; index=61
                                                                at com.felhr.usbserial.FTDISerialDevice$FTDIUtilities.copyData(FTDISerialDevice.java:594)
                                                                at com.felhr.usbserial.FTDISerialDevice$FTDIUtilities.adaptArray(FTDISerialDevice.java:502)
                                                                at com.felhr.usbserial.UsbSerialDevice$WorkerThread.run(UsbSerialDevice.java:247)

Has anyone else had success or issues with this adapter? Other multi-port adatpers?

Link to StartTech

https://www.startech.com/uk/Cards-Adapters/Serial-Cards-Adapters/USB-to-RS-232-2-Port-Serial-DB9-Adapter~ICUSB232C2

bill-chadwick, 2017-11-13:
I should perhaps add that this is using Android 7 on an S6 phone.

bill-chadwick, 2017-11-13:
I can get both ports working but only at slower baud rates e.g. 38400 rather than 115200. Might be worth seeing if setting the priority of the UsbSerialDevice worker thread to Thread.MAX_PRIORITY helps. Or it could just be that the Startech dual port device is not very good (is dropping bytes at high speeds).

bill-chadwick, 2017-11-13:
Device is actually 

https://www.startech.com/uk/Cards-Adapters/Serial-Cards-Adapters/2-Port-USB-RS232-Serial-Cable~ICUSB2322F

felHR85, 2018-10-28:
Hi @bill-chadwick 
Actually I've not tried UsbSerial with any of them.
Did you finally solve it or got any clue?

bill-chadwick, 2018-10-30:
I never got it to work but have not tried again since my original post.

closed 2018-12-31 by felHR85