felHR85/UsbSerial #131: Some commands sent over OTG are lost

Srishesh, 2017-12-06:
I have been using this library for sending commands over OTG. It is working perfectly but sometimes the commands that we send are missing. Like out of 10 commands, 3 commands are lost and does not reach the device. It was demonstrated that the application couldnt send sometimes.

closed 2018-10-28 by felHR85

dipen30051989, 2018-11-14:
I'm facing the similar issue. Some time app is not able to send command over USB. Command passed to the write function but some how library is not able to send command over USB. I tried to debugging the issue by putting the USB analyser. On the analyser it's not showing any command pass through USB from android. Let me know if is there any solution for that.

felHR85, 2018-11-14:
@dipenpatel30 More information would be great:
- Android version.
- Usb to serial adapter.
- Phone model.
- Code example
...