felHR85/UsbSerial #132: Is there a way to reveive multiple byte   data  like float and int64

Aerotic, 2017-12-15:
As the title, I'm trying to receive data consisting of several bytes. But I've found that if I send a float then the phone receive data with different length from 1 to 8 bytes....

RicoLewis, 2017-12-21:
Hello, No you can't with one byte, 8-bit are from 0 to 255 in Decimal, 00 to FF in Hexadecimal, 00000000 to 11111111 in binary. 7-bit are from 0 to 127 etc.
You can send a macro of bytes but your phone (or receptor) need to know read it.
If you want send 1.234, the ASCII code decimal will be : 49 46 50 51 53
49 for 1, 46 for . etc.
So your phone receive these 4 bytes.

Aerotic, 2017-12-22:
@RicoLewis Thanks for your answer.  But I've tried a new way to receive float data through serial port and success.
Since the length of each received data frame is fixed, I set a counter to record the number of bytes which the phone has received. If the counter reaches the length, I'll set the counter to zero and read float from those bytes.
However, there's still a risk. If the serial port receives an unexpected byte, all of the data will be stored in wrong positions and the float value read from those bytes will be wrong.

eyedol, 2018-01-13:
I'm going to close this since it's a question rather an issue.

closed 2018-01-13 by eyedol