felHR85/UsbSerial #142: ArrayIndexOutOfBoundsException From PL2303SerialDevice

eyedol, 2018-01-08:
Here's the relevant stacktrace:

```java
Caused by java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
       at android.hardware.usb.UsbDevice.getInterface(UsbDevice.java:232)
       at com.felhr.usbserial.PL2303SerialDevice.<init>(PL2303SerialDevice.java:54)
       at com.felhr.usbserial.UsbSerialDevice.createUsbSerialDevice(UsbSerialDevice.java:76)
       at com.felhr.usbserial.UsbSerialDevice.createUsbSerialDevice(UsbSerialDevice.java:58)
```

eyedol, 2018-01-08:
This issue seems to be related to [this one](https://issuetracker.google.com/issues/37032363). Then that looks like an OS level bug

felHR85, 2018-01-08:
Another of those beautiful Android USB related bugs. I will look at it next weekend to be if some fix can be implemented.

lordjabez, 2018-01-22:
I just ran into this bug today as well, on Android 5.1. My project's at a standstill until I can figure this one out. I might take a crack at a workaround if you haven't had a chance yet, @felHR85 .

felHR85, 2018-01-22:
Sure! @lordjabez It would be great.

lordjabez, 2018-01-22:
I did some reading today, and the fixes so far are all within the Android code itself, not the application side (see the link @eyedol posted). Rebuilding/upgrading the OS isn't an option for me, do you have any thoughts on what could be done app-side?

This is the best idea someone in the above thread had:

> The only privileged bit in the USB API is the usbfd file descriptor from the service.  It might be possible to work around this with JNI or reflection to create UsbEndpoint objects (there's nothing magic in there).  You can discover the actual interfaces and endpoints by trawling /sys directly. 
> 
> It would be ugly and not API portable and a lot of extra work, but if you're desperate for a workaround it would probably function.

felHR85, 2018-01-28:
I have been investigating this issue and it is basically not possible to solve this from android user-space. Sadly I think the only way is not using Android 5.1 if possible.

eyedol, 2018-01-29:
I added a note #145 to the readme regarding this issue. Essentially recommending against the use of `Android 5.1.1` due to this bug. 

I was thinking we drop support for `Android 5.1.1`. Basically, bump the minSDK from `12` to say `23` but looking at the platform versions [distribution chart](https://developer.android.com/about/dashboards/index.html), `5.1.1` seems to be widely used. `19.4%` is a substantial amount in the Android world. Thoughts?

eyedol, 2018-01-29:
> You can discover the actual interfaces and endpoints by trawling /sys directly.
@lordjabez or @felHR85, might you know what this entails?

lordjabez, 2018-01-29:
@eyedol I only know enough about it to know I couldn't actually get it to work, and if somehow it did work it wouldn't be nearly robust enough to recommend to anyone else.

felHR85, 2018-01-29:
@eyedol @lordjabez The only way I see would be using Android NDK, JNI and C/C++ code taking advantage that PL2303, CP2102 and all of those drivers have their drivers implemented in the linux kernel that powers Android.
I attach here a little snippet that I've used in the past to open a serial connection. This code, correctly glued through JNI and compiled with the NDK should work in **at least any Android rooted device**
```c
int init_serial()
{
  fd_serial_port = open("/dev/ttyS1", O_RDWR); // Must be open in  blocking mode 

  if(fd_serial_port == -1)
    return -1;
    
  /*
     Configure Termios
  */
  struct termios current_config; 
  tcgetattr(fd_serial_port, &old_config);
  current_config = old_config;

  current_config.c_iflag = 0;
  current_config.c_oflag = 0;
  current_config.c_cflag = CS8 | CREAD | CLOCAL;
  current_config.c_lflag = 0;
  current_config.c_cc[VMIN] = 1;
  current_config.c_cc[VTIME] = 0;

  // Baud rate config
  cfsetispeed(&current_config, B115200);
  cfsetospeed(&current_config, B115200);

  // Set configuration
  tcsetattr(fd_serial_port, TCSANOW, &current_config);
  
  is_initialized = 1;
  return fd_serial_port;    
}
```
This method returns a file descriptor if the connection has been opened properly. Writing should be simple using [write(2)](http://man7.org/linux/man-pages/man2/write.2.html) and [read(2)](http://man7.org/linux/man-pages/man2/read.2.html).

But I think this is overly complicated (most because of NDK and JNI than the C code itself) so I discourage anybody to try this approach.

closed 2018-11-03 by felHR85

ltd9938, 2018-11-07:
I know I'm bumping this (sorry), however my team is ordering a batch of tablets running 5.1 for our app.

What exactly raises this exception? Does this bug render UsbSerial useless on 5.1? 

Our app is just a simple interface that sends basic serial commands so it isn't too complex.

lordjabez, 2018-11-07:
UsbSerial can be used, but it's a pain. Only the last USB device plugged in to the tablet will be functional.

felHR85, 2018-11-07:
@ltd9938 @lordjabez  I understand it should work normally if you only plug one device right?

lordjabez, 2018-11-08:
Probably, but if you restart the tablet and don't unplug and re-plug in the device it might not work until you do. The enumeration order of USB devices can be somewhat arbitrary.

ltd9938, 2018-11-08:
Thanks for the info guys. We're probably just going to avoid it all together.

eyedol, 2018-11-09:
@ltd9938 I'm not sure how fixed of a constraint you've to use Android 5.1 but my team and I have successfully used it on Android 6.0.1 Marshmallow with two USB devices. If you're able to upgrade to a later version of Android you won't have this particular issue.

mtinnes, 2018-11-30:
Anyone know if the CP2102SerialDevice driver has the same issue? We are stuck with 5.1.1 and are getting the same exception on occasion. The device fails to be detected from a cold start, unplugging-replugging the device seems to be the only way to properly detect and use the device.

felHR85, 2018-12-08:
@mtinnes As far as We know there is no way to solve this in 5.1.1

midnight-wonderer, 2018-12-11:
Also, can we catch `ArrayIndexOutOfBoundsException` around `device.getInterface` and raise a different exception specific to this library?

fah33mkhan, 2019-04-24:
I have used https://github.com/mik3y/usb-serial-for-android for connecting for PL2303 my problem is that connection is successful but if there is nothing to read and write for sometime 15-20 sec then I am unable to read or write further, The exception is raised sometimes but sometimes it is successful any help, please?

eclectice, 2019-04-25:
@fah33mkhan If you are using https://github.com/mik3y/usb-serial-for-android
as an alternative USB-SERIAL library, you need to use another fork by
kai-morich who is the current maintainer of that library. In that case,
please try his real-world sample app
https://github.com/kai-morich/SimpleUsbTerminal and his forked library
https://github.com/kai-morich/usb-serial-for-android