felHR85/UsbSerial #158: CH340 only sends the characters to the callback after the USB cable is disconnected.

tiagotobias2003, 2018-08-01:
Hello...
I'm using the library in asynchronous mode, everything works perfectly when connected to the ATmega16u2 controller on the genuine Arduino board, communicating with Android 7.0 Nougat.

However, when the same system runs using generic Arduino, with CH340 to control USB, I realize that the data is no longer sent to `mActivity.get().display.append (data);` the characters only appear on the display when the USB cable is disconnected.

Did someone else also have a problem like that?

mengguang, 2018-09-05:
I have the same problem.

jacky830, 2018-09-28:
I have the same issue; anyone have suggestions on how to solve it?

zjn0505, 2018-09-28:
For me it only happens on baud rate 9600.

RaudByorn, 2019-02-20:
May be the problem is that you sending more bytes, than buffer size is. Try to use SerialnputStream to use a custom buffer size